bertmelis/VitoWiFi #3: readDatapoint.ino crashes on ESP32

Empor-co, 2017-09-25:
Hi,

I tried to run the example on a Adafruit HUZZAH32 – ESP32 Feather Board and switched Serial and Serial1 for better debugging.
Sadly the example keeps crashing when calling VitoWifi.loop(); in the loop.

Any ideas?

Thank you in advance.

Best regards,

E-co

bertmelis, 2017-09-25:
I'll check this evening (only havze a Wemos Lolin32 at my disposal, but the behaviour should be the same). To be honest, I only have tested the esp8266 which is working 
Is there any crash info? (stack dump)

bertmelis, 2017-09-25:
Could it be that I didn't mind the pin mappings in the examples?
I didn't specify any pins in the examples to avoid `ifdef`-clutter. But it renders the examples unusable on the ESP32.
Furthermore, I also didn't implement the pin mapping in the Optolink.
I can override `VitowifiClass::setup(HardwareSerial* serial)` with `VitowifiClass::setup(HardwareSerial* serial, int8_t rxPin, int8_t txPin)` and subsequently override `Optolink::begin(HardwareSerial* serial);` with `Optolink::begin(HardwareSerial* serial, int8_t rxPin, int8_t txPin)`.

Empor-co, 2017-09-25:
Shouldn't this be part of the hardware serial lib?
(https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp)

bertmelis, 2017-09-25:
On the Huzzah, you can use RX and TX for the pins as these are defined but Serial1 is not defined/initialized by default.

Empor-co, 2017-09-25:
This is my current setup:
[image]

bertmelis, 2017-09-25:
could you test with the test-esp32 branch?
I've only tested myself using the globalCallback-esp32 example.

Empor-co, 2017-09-26:
Compiles nicely and doesn't crash.
Adjusted   VitoWifi.setup(&Serial1, 16, 17); to match 16=RX and 17=TX.
Will check it with the Vito tomorrow.

bertmelis, 2017-09-26:
I tested globalCallback-esp32 myself with a Wemos Lolin32 on a Vitodens 200, with Vitotronic 200, type HO1. I'd love to hear your feedback.
Main developement is on a esp8266 as I try to keep it online for stability testing (which is in doubt for the moment).

Empor-co, 2017-09-27:
The code seems to be stable now - just my board setup seems to be wrong: None of the codes I sent, are recognized. Maybe I got the wrong transmit diode.....

bertmelis, 2017-09-27:
make sure there's no "optical shortcut" and that the optics are aligned. I'll look into the types of parts I have.

bertmelis, 2017-09-27:
I use a SFH309FA (24°) transistor (with black body!) and a SFH487-2 led, both 3mm versions. I followed this guide: http://openv.wikispaces.com/Bauanleitung+ESP8266.

Empor-co, 2017-09-28:
That link was the base of my setup the SFH487-2 seems to be end-of-sales, but I was able to get some at ebay. Will try to better seperate the diodes and better shield the vito receiver from any disturbing light....work in progress

Empor-co, 2017-09-30:
[image]
Tried to build a cap to prevent the optical shortcut, but no luck so far.

bertmelis, 2017-09-30:
where does it hang? Does it stay in the "reset" state?

I'm working on the docs with pictures from my setup.

You could also try to power up the heating (toggle mains power) while the optocoupler is attached. On mine, the heating won't start. It's a sign that the coupler works.

Empor-co, 2017-09-30:
Yes, reset state.
Heating starts are usual.
I guess I have to wait for the other LEDs I ordered... ;-)

Empor-co, 2017-10-03:
One step further: my heating doesn't use the 300 protocol, but the KW protocol, which also starts with a heartbeat of 5s, but is stateless. Have to figure out how to "synchronize" to the heartbeat in order to destinguish the heartbeat from the answer...

bertmelis, 2017-10-03:
The KW protocol is on my to do-list.
Closing this one in favor to #4

closed 2017-10-03 by bertmelis