bertmelis/VitoWiFi #4: Implement KW protocol

bertmelis, 2017-10-03:


Empor-co, 2017-10-03:
BTW: The other LED diodes arrived: the Vishay TSHA4401 (IR LED 875nm) seem to work fine too - since the SFH487 family seems to be end-of-sales.

bertmelis, 2017-10-03:
Developement takes places in [the develop-KW branch](https://github.com/bertmelis/VitoWifi/tree/develop-KW). I made a quick first commit but it is completely untested!

Empor-co, 2017-10-03:
Adjusted the datapoints and TX RX:
`  VitoWifi.addDatapoint("outsidetemp", "boiler", 0x0800, TEMP);
  VitoWifi.addDatapoint("boilertemp",  "boiler", 0x0810, TEMP);
  VitoWifi.setGlobalCallback(globalCallbackHandler);
  VitoWifi.setLoggingPrinter(&Serial);
  VitoWifi.enableLogger();
  VitoWifi.setup(&Serial1, 16, 17);`

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0010,len:4

load:0x3fff0014,len:588

load:0x40078000,len:0

load:0x40078000,len:9880

entry 0x400789d8

Setup finished...

reading datapoints

Datapoint outsidetemp READ action added

Datapoint boilertemp READ action added

READ request on address 0008, length 2

succes

Datapoint outsidetemp action successful

Value received

boiler - outsidetemp: 13.6

READ request on address 1008, length 2

succes

Datapoint boilertemp action successful

Value received

boiler - boilertemp: 43.2

READ request on address 1008, length 2

succes

Datapoint Guru Meditation Error of type LoadProhibited occurred on core  1. Exce                                                                                                                                                                             ption was unhandled.

Register dump:

PC      : 0x400e4af3  PS      : 0x00060530  A0      : 0x800d106f  A1      : 0x3f                                                                                                                                                                             fca820

A2      : 0xfefefefe  A3      : 0x3f40103f  A4      : 0x00060120  A5      : 0x00                                                                                                                                                                             000001

A6      : 0x00060120  A7      : 0x3ffc8920  A8      : 0x800d1938  A9      : 0x3f                                                                                                                                                                             fca800

A10     : 0x400d17d4  A11     : 0x3f40103f  A12     : 0x80083f1c  A13     : 0x3f                                                                                                                                                                             fc0bb0

A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000018  EXCCAUSE: 0x00                                                                                                                                                                             00001c

EXCVADDR: 0xfefeff02  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xff                                                                                                                                                                             fffffd

 

Backtrace: 0x400e4af3:0x3ffca820 0x400d106c:0x3ffca840 0x400d09a4:0x3ffca870 0x4                                                                                                                                                                             00ddda1:0x3ffca890

 

Rebooting...

ets Jun  8 2016 00:22:57

 

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0010,len:4

load:0x3fff0014,len:588

load:0x40078000,len:0

load:0x40078000,len:9880

entry 0x400789d8

Setup finished...

READ request on address 0000, length 0

succes

Datapoint Guru Meditation Error of type LoadProhibited occurred on core  1. Exce                                                                                                                                                                             ption was unhandled.

Register dump:

PC      : 0x400e4af3  PS      : 0x00060530  A0      : 0x800d106f  A1      : 0x3f                                                                                                                                                                             fca820

A2      : 0xfefefefe  A3      : 0x3f40103f  A4      : 0x00060120  A5      : 0x00                                                                                                                                                                             000001

A6      : 0x00060120  A7      : 0x00000000  A8      : 0x800d1938  A9      : 0x3f                                                                                                                                                                             fca800

A10     : 0x400d17d4  A11     : 0x3f40103f  A12     : 0x8008398c  A13     : 0x3f                                                                                                                                                                             fca720

A14     : 0x00000000  A15     : 0x3ffca7f0  SAR     : 0x0000001f  EXCCAUSE: 0x00                                                                                                                                                                             00001c

EXCVADDR: 0xfefeff02  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xff                                                                                                                                                                             fffffd

 

Backtrace: 0x400e4af3:0x3ffca820 0x400d106c:0x3ffca840 0x400d09a4:0x3ffca870 0x4                                                                                                                                                                             00ddda1:0x3ffca890

 

Rebooting...

ets Jun  8 2016 00:22:57`

bertmelis, 2017-10-03:
I just tested myself with an esp8266 and it doesn't work at all :cry: 
I have to redo the complete optolink FSM I guess.

The addresses is something strange. Endianess? Is the ESP32 different from the ESP8266 wrt endianess?

Empor-co, 2017-10-04:
the crazy part is that it worked in the first iteration. but yes, looks like the msb block moved

bertmelis, 2017-10-09:
Could you try again with the newest commit?
I'm not completely if the "reset to KW"-method will also work on devices not compatible with the 300-protocol.

Empor-co, 2017-10-11:
no luck so far: "Setup finished.." is displayed all the time, but no readings...

bertmelis, 2017-10-11:
You're using the globalcallback-esp32?
I must have changed something on the serial-initialisation then... If you don't even see the "reading datapoints" message, we're back at where we started.

I can only test myself this evening.

bertmelis, 2017-10-12:
I had the #ifdef-guard for esp8266 removed for some reason but it keeps the esp32 from starting up. I just committed the fix after testing with my lolin32 board.

Empor-co, 2017-10-12:
I'll be offline for about 3 weeks :/ will test it directly as soon as I'm back.

bertmelis, 2017-11-10:
If you want, you can also test with the branch `develop-merge_protocols`

You'll have to add `VitoWifi_setProtocol(KW);` to your sketch, before `setup()`.

Empor-co, 2017-11-11:
Will try to find some time this weekend to pick-up testing...

bertmelis, 2017-11-14:
Deleted the KW-branch as you can use the develop-branch and set the KW-protocol in your sketch.

Empor-co, 2017-11-21:
Just tested the dev-branch and it works nicely!
Just found some typos in OptolinkKW, line 163 & 164 as well as optolinkp300, line 310, 31
success with two 's' at the end ;-)

closed 2017-12-30 by bertmelis