bertmelis/VitoWiFi #12: Platform independent library

saidlm, 2017-11-24:
Hi,
Is there any possibility to make the code independent from hardware platform (ESP chip)? I am about to built system based on Arduinos as let me say protocol converters and RPi with Domotics as user facing gateway connected to Vitocal 200s via Optolink and Vitosolic 200 via vbus to be able to monitor and configure whole system remotely. I prefer to use wired solution instead of WiFi (because of security, availability and stability). In fact from my point of view there is no dependency to WiFi and ESP chip and to have the the library platform independent is not bad idea in my opinion.
 
 I have made some changes in source code and have tried to compile it for Arduino Nano board but have no success. Honestly speaking unfortunately I am not so good in C++ :-(.  

Thank you

bertmelis, 2017-11-24:
I can surely try. The code is largely generic c++. 

However, some newer c++ features are not available in the Arduino compiler.

Do you have a particular board in mind to start with?

saidlm, 2017-11-27:
My idea i to use your code together with MySensors library. I would like to build up sensor network based on RS-485 bus. I have plan to use Arduino Nano as a Gateway and Arduino Mini Pro as sensors. 

BTW: I have attached documentation for Vitotronic 200 wo1 which is missing in your documentation folder now.

[vt200wo1c_en.pdf](https://github.com/bertmelis/VitoWifi/files/1505914/vt200wo1c_en.pdf)
[vt200wo1ab_wpr300en.pdf](https://github.com/bertmelis/VitoWifi/files/1505915/vt200wo1ab_wpr300en.pdf)

bertmelis, 2017-12-01:
So you want to have VitoWifi ported to a Arduino Mini Pro?

Keep in mind that the ESP's have way more RAM than the Arduino's. It's 100kB vs 2kB. I doubt I can make VitoWifi compatible with the AVR-boards. It's probably doable to create a slim version with only the optolink itself and not the complete VitoWifi dataholding-structure and interface.

Anyway, I'm not familiar with Arduino's. I'm only familiar with the Arduino framework for ESP8266 and ESP32. Furthermore, I don't have such a board for testing so I probably can make the code compile but testing will have to be done by you.

Maybe I also create a seperate repo/fork, as this repo is named "VitoWifi" and when you're using a wired connection... it's a bit of an odd name 😄 

What do you think?

PS I only learned C++ 1 year ago. On my own.

saidlm, 2017-12-05:
You are right. It is exactly approach which I have decide to follow. I have made a minority changes in your code dealing only with optical interface. I started with P300 but finally I have recognized that P300 is not supported by Vitotronic 200 wo1c. Which was bad surprise for me :-(. I had to switch to KW protocol and finally made the communication established.

The major respectively the only change which I had to do was to change definition of begin function (...#ifdef ESP8266...).  After that I was able to use this part of code as part of my sketch. Unfortunately it is not possible to make the changes in library tree and to include it from there. It somehow tries to compile all the library and it ends with error.

The second issue is very pure Viessmann documentation :-( It is almost impossible to use the document which I have uploaded above. I spent so much time to discover right addresses for all temperature sensors. After about 2 hours experimenting with it I was more less successful. 

The next good thing is that MySensor network base on RPi with USB<->RS485 converter and Arduino Pro Mini with SN75176 as RS485 driver is up and running.

Currently I am about to start to work on code which will realize the sensor for Vitocal 200-s. I will share the code on GitHub of-course as soon as it is in some working release. 

In mean time I have done physical part of the project - optical connector. It is designed in OpenSCAD and printed on my 3D priter. All required components are published here: https://github.com/saidlm/Viessmann-optical-interface.

P.S. I am from old school ;-). I started programming in assembler on Z-80, then HC11, PIC16F84 etc. I have still problem to see world in object manner... ;-)