bertmelis/VitoWiFi #16: 4byte values KW protocol

siggiull, 2018-01-31:
Heating type: 2098 -  Vitotronic 200 KW2
Protocol: KW
Board: ESP8266 NodeMCU
#################################
Hello Bert,
First of all I want to thank you for your work on this library, really cool stuff!

I have a problem reading 4byte values with the KW protocol. As soon as the first datapoint with 4 bytes is read, it returns ERROR -1, and all following readings fail, too.
My first thought about this is, could it be that " uint8_t _rcvBuffer[2]; " in "OptolinkKW.h" should be [4]?
I check that asap.

Siggi

bertmelis, 2018-01-31:
I understood from the Openv Wiki that the KW protocol supports datapoints up to 2 bytes. So It actually never implemented reading 4 bytes.

I can be wrong as I'm not a native German speaker.

siggiull, 2018-01-31:
What they say is that the KW protocol supports addresses with 2byte. Data could be up to 4 (as far as I know).
I checked it today, with the array set to the length of 4 it works.
While testing longer today I often had the problem that the first "round-trip" of reading the data points was ok, the next one every data point gets wrong data. I will do further investigation on this one.

bertmelis, 2018-02-01:
I see, my fault. I'm a bit short on time but will look into this. Error handling is buggy.

bertmelis, 2018-02-09:
Apologies for the delay. Free time...sigh.

Changing the buffer size from 2 to 4 is probably doing the trinck as you already suggested. There are some `memset`s but that is more for debug purposes and are not really needed when everything's working properly.

Error handling needs to be refactored (or rebuild from ground up?).

siggiull, 2018-02-10:
Free time is always the bottleneck :(
Right now I'm fighting with multiple problems, starting with homie always starts in config mode etc.
And I don't now if I have pysical problems with the serial connection. If there some news to share, I let you know.

bertmelis, 2018-02-10:
my experience: I once completely erased my esp with espressif's esptool. after that I could upload to config without any problems.

siggiull, 2018-02-10:
Thank you, I will try that.

bertmelis, 2018-04-24:
Closing as issue should be resolved with commit f6257085c342971f4fc65e1d86a89f1edbe0a765.

Feel free to reopen if needed!

closed 2018-04-24 by bertmelis