bertmelis/VitoWiFi #19: "Listen only" mode

Jojo-A, 2018-02-16:
Dear Bert,
thanks a ton for your effort and this project!

I currently have a Vitoconnect connected to my heating control unit. Now I like to add a DIY adapter to the OPTOLINK interface to make the heating data available to my home automation system (it is ridiculous, that Viessmann is not planning to provide a public API for their Vitoconnect >:( ).

To not disturb/the "normal" communication between Vitoconnect and the control unit, I like to firstly analyze the communication between both.
- what is send/requested by the Vitoconnect
- what are the answers of the control unit
- what is the timing/intervals of communication between both

With the knowledge about these things, I could decide, if it is necessary to transmit/request data by myself, or if it is sufficient to just process the data, that is requested by the Vitoconnect anyway.

Now the question: can I use your software for the ESP8266 to _just listen_ to the communication that is going on? I would define the protocol, and then the software should just process all incoming data and decode it according to the defined protocol.

What do you think about this approach?

Thanks and greetings

bertmelis, 2018-02-16:
I also think it's very outdated not to have a public API. I strongly believe they would sell much more Vitoconnect systems because people won't be tempted to build DIY solutions. Now it only leads to fragmentation of the IoT landscape.

Anyway, obviously VitoWifi was not designed to hack into an existing connection like VitoConnect. But I like the idea.
By just sniffing and analyzing the traffic you can filter out "answer-messages": they start with `0x41 07`. Luckily the address of the datapoint is sent along with the value. Transmitting yourself I would not recommend. I read somewhere that it would make the connection(s) unstable --> both connections receive data they don't expect or even collide.

But it is out of the scope of this repo. Furthermore, I don't have a VitoConnect so it's quite hard for me to work on this feature.

bertmelis, 2018-02-19:
Implementing this in the current base will take a redesign of the FSM.
I'm also not shure how to do this. Because VitoWifi can not know when or what it is receiving in a listen-only" state. So on receive, an iteration in all the known DPs has to be done to launch the correct callback.
Another solution could be to launch a general callback with the DP, length and payload. Then it's up to you to do the correct translation and do something useful.

(Meanwhile, I try to contact Viessmann to consider implementing a public API theirselves)

Jojo-A, 2018-02-20:
@bertmelis hey, thank you for looking into this :) .
my idea was indeed, to just iterate through all known datapoints to determine what has been received. This is why the protocol has to be defined in my model. Otherwise it would get even more complicated to first determine the protocol and then the actual datapoint...
But of course I understand when you say, that this is not possible with the current code base. No problem at all.
About the API: actually I am not very confident about that. I already asked them publically and then answered, that it is "not implemented and not planned". Totally ridiculous, that they force my to blast my data once around the world, then just make these data accessible locally. It is about _my_ data in _my_ WiFi!

bertmelis, 2023-12-08:
Closing in favour of #87

closed 2023-12-08 by bertmelis