bertmelis/VitoWiFi #25: Linker error in Example readDatapoint: Logger

s0170071, 2018-03-31:
### Installation specifics

* **Board:**  Wemos D1 mini.

### Symptom
```
Linking everything together...
"/home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -g -w -Os -nostdlib -Wl,--no-check-sections -u call_user_start -u _printf_float -u _scanf_float -Wl,-static "-L/home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/tools/sdk/lib" "-L/home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/tools/sdk/ld" "-L/home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.4m1m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read  -o "/tmp/arduino_build_6867/readDatapoint.ino.elf" -Wl,--start-group "/tmp/arduino_build_6867/sketch/readDatapoint.ino.cpp.o" "/tmp/arduino_build_6867/libraries/VitoWifi/Datapoint.cpp.o" "/tmp/arduino_build_6867/libraries/VitoWifi/OptolinkKW.cpp.o" "/tmp/arduino_build_6867/libraries/VitoWifi/OptolinkP300.cpp.o" "/tmp/arduino_build_6867/libraries/VitoWifi/VitoWifi.cpp.o" "/tmp/arduino_build_6867/arduino.ar" -lhal -lphy -lpp -lnet80211 -llwip2 -lwpa -lcrypto -lmain -lwps -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group  "-L/tmp/arduino_build_6867"
/tmp/arduino_build_6867/libraries/VitoWifi/OptolinkP300.cpp.o: In function `OptolinkP300::_printHex(Print*, unsigned char*, unsigned char)':
/home/john/Arduino/scetchbooks/libraries/VitoWifi/OptolinkP300.cpp:407: undefined reference to `Logger::Logger()'
/tmp/arduino_build_6867/libraries/VitoWifi/OptolinkP300.cpp.o: In function `OptolinkP300::OptolinkP300()':
/home/john/Arduino/scetchbooks/libraries/VitoWifi/OptolinkP300.cpp:42: undefined reference to `Logger::Logger()'
/tmp/arduino_build_6867/libraries/VitoWifi/VitoWifi.cpp.o: In function `swap<unsigned int>':
/home/john/Arduino/scetchbooks/libraries/VitoWifi/VitoWifi.cpp:266: undefined reference to `Logger::Logger()'
collect2: error: ld returned 1 exit status
Using library VitoWifi in folder: /home/john/Arduino/scetchbooks/libraries/VitoWifi (legacy)
exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

s0170071, 2018-03-31:
Ah, I should add that I am using the Arduino IDE. I heard that Platformio handles includes differently....

bertmelis, 2018-04-01:
I removed the Logger class and inlined all the logging functions. So this should be solved in the next version. I'm testing some things before I make the push though.

bertmelis, 2018-04-03:
#31

closed 2018-04-03 by bertmelis