bertmelis/VitoWiFi #27: Feature request: Software serial

s0170071, 2018-04-01:
I know, there is nothing like a UART. But since the ESP8266 only has one developing with it is... tricky.
The same pins are used for communicating with Vito, for debug and programming.

The Vitostuff does communicate so slow that a software serial would be a real option. And the Vito connection could be moved somewhere else. 

Alternative: Serial.swap() to use TX2/RX2. You can swap at run-time and I 've seen programs doing that frequently just for debug. Just mind to flush() before swapping.

bertmelis, 2018-04-01:
I used to have SS, but I removed it didn't seem useful.

Furthermore, the standard library doesn't support 8E2 mode. So it only worked with a "quick and dirty" modified version.

If you really want I van give you some hints but I'm reluctant to merge it into the main codebase.

s0170071, 2018-04-02:
What you could do is  Serial.swap()  TX0/RX0 to TX2/RX2 and connect the Vito here. Then
 
a) then wire TxD1 to TxD0 and debug via TX1.
b) then use soft UART on TxD0 and debug via soft uart.

I use a) now, works fine.
When you connect the Vito to TX2, don't load the pin with the IR transmitter directly. Use a transistor. The pin changes states during boot up. If you tie it hard to vcc, the ESP doesn't boot.

closed 2018-04-03 by s0170071