bertmelis/VitoWiFi #28: Exception 28

s0170071, 2018-04-01:
```
void setup() {
  // setup VitoWifi using a global callback handler
  VitoWifi.addDatapoint("outsidetemp", "boiler", 0x5525,STAT);
  VitoWifi.addDatapoint("boilertemp", "boiler", 0xf802, STAT);
  VitoWifi.setGlobalCallback(globalCallbackHandler);
  VitoWifi.setup(&Serial);
  
  // setup VitoWifi using a global callback handler
  VitoWifi.enableLogger();
  VitoWifi.setLogger(&Serial1);

  Serial1.begin(115200);
  Serial1.println(F("Setup finished..."));
}

void loop() {
  static unsigned long lastMillis = 0;
  if (millis() - lastMillis > 60 * 1000UL) {  // read all values every 60 seconds
    lastMillis = millis();
    VitoWifi.readDatapoint("outsidetemp");
    VitoWifi.readDatapoint("boilertemp");
  }

  VitoWifi.loop();
}
```
crashes. Exception decoder shows: 

Decoding 10 results
0x402033e4: VitoWifiInterface ::loop() at /home/john/Arduino/scetchbooks/libraries/VitoWifi/VitoWifi.cpp line 266
0x40203a41: HardwareSerial::available() at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.cpp line 188
0x40204148: OptolinkP300::_clearInputBuffer() at /home/john/Arduino/scetchbooks/libraries/VitoWifi/OptolinkP300.cpp line 376
0x402040f0: Logger::write(unsigned char) at /home/john/Arduino/scetchbooks/libraries/VitoWifi/Logger.cpp line 41
0x40204240: Print::print(__FlashStringHelper const*) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 101
0x402033d8: std::queue    > >::front() at /home/john/Arduino/scetchbooks/libraries/VitoWifi/VitoWifi.cpp line 266
:  (inlined by) VitoWifiInterface ::loop() at /home/john/Arduino/scetchbooks/libraries/VitoWifi/VitoWifi.cpp line 196
0x401068ee: millis at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c line 183
0x4020268e: loop at /home/john/Arduino/github/VitoWifi/examples/readDatapoint/readDatapoint.ino line 47
0x40203f1c: loop_wrapper at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 58
0x40100739: cont_wrapper at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 81

s0170071, 2018-04-01:
EDIT: It may have something to do with changing the datatype to STAT.

s0170071, 2018-04-01:
in witowifi.cpp function(s) loop, right after if (_optolink.available() > 0) 
Seems like the queue is empty and still you're trying to print out the dp's name. What happens if you access an empty queue ? 
I exited the function if the queue is empty, seems to fix the crash. Communications is still not working though....

bertmelis, 2018-04-01:
Why would you change a TEMP to STAT? Your Vitotronic should give an error.

Anyway, accessing an empty queue results in a crash obviously, with error 28. I'm looking into this as error handling is buggy.

s0170071, 2018-04-01:
Actually it does give an error, I only didn't notice it because of the crash. 
I changed another thing in the receive handler, right after   if (_rcvBuffer[0] != 0x41) return;  
 
```
  if (_rcvBufferLen >1) {
    _rcvLen= _rcvBuffer[1] + 3; // 3 because 0x41, len, crc
    }  

```
This checks if the message length has received and uses that info for the remaining bytes rather than the expected size.  Prevents you from timing out if you use a wrong data type.

Why I use a wrong data type? I do simply not have a list of data points and their types. The list on http://openv.wikispaces.com/Adressen does not state the type of the returned data. That is more or less the root of all evil here... Me trying the wrong addresses together the wrong data types...

Since I am probably not the only fool in the world, maybe returned data can be auto-interpreted from the number of received bytes and then be cast to double ? Or do you have another list of data points ?

s0170071, 2018-04-01:
Ah forgot, I got it communicating now and hooked it up to MQTT. My code still needs some beautification but not now, let's call it a day....

If you're interested though, let me know ;-)

bertmelis, 2018-04-01:
Code additions/pull requests are always welcome!

Wrt the datapoints: you can indeed do some magic casting, but then you still won't have the correct data as there are conversion factors to deal with. But I may have a way to get you your list. I'll check tomorrow...

s0170071, 2018-04-03:
See #29

closed 2018-04-03 by s0170071