bertmelis/VitoWiFi #30: Feature request: value factor

s0170071, 2018-04-03:
Some datapoints require applying a factor to the reading. How about adding that factor to 
VitoWifi.addDatapoint(..., factor) so that the globalCallbackHandler has that factor or the correct value available ?

bertmelis, 2018-04-03:
My idea was to handle the conversions by the different datapoint-types. So the average user doesn't have to deal with the lower level bits and bytes.

This way, the main program can expect `26,3` when reading `outsideTemperature` instead of `0x0107`.

s0170071, 2018-04-04:
Then I seem to have a new one. My "JAZ" reading, now being read as 
`VitoWifi.addDatapoint("JAZHeizen", "boiler", 0x1681, MODE);`
needs to be divided by 10.
Reading now is 49, it should be 4.9

bertmelis, 2018-04-04:
So it is a temperature, 1 byte, unsigned, factor 10?

s0170071, 2018-04-04:
No temperature. Efficiency factor. Not sure about the number of bytes.

bertmelis, 2018-04-04:
aha, COP (coefficient of performance). Will implement this evening if time allows.

bertmelis, 2018-04-04:
Please check with [COP-datapoint branch](https://github.com/bertmelis/VitoWifi/tree/COP-datapoint).

s0170071, 2018-04-04:
Will do. Give me some days. Thanks.

s0170071, 2018-04-11:
The data point is working. The transformation is OK, but there is an issue with the nack. When it hits its first nack, all subsequent reads fail. I think I debugged this already. Let me look...

In the meantime , here is the log.

```

boottt..

Seri⸮
Connecting to MNET
.....
WiFi connected
IP address: 
192.168.1.38
Setup finished...
Attempting MQTT connection...connected
READ boilertemp
READ outsidetemp
READ vorlauftemperatur
READ Speichertemperatur
READ RaumtemperaturSoll
READ RedRaumtemperaturSoll
READ PartyRaumtemperaturSoll
READ HeizkennlinieNiveau
READ HeizkennlinieSteigung
READ JAZgesamt
READ JAZHeizen
READ JAZWW
READ COPHeizen
READ COPWW
READ EinschaltungenVerdichter
READ 41050001010d0216
ack
RCV 41070101010d02e60100
ack
DP boilertemp succes
boiler - boilertemp: 48.6
READ 410500010101020a
ack
RCV 41070101010102a300b0
ack
DP outsidetemp succes
boiler - outsidetemp: 16.3
READ 41050001010a0213
ack
RCV 41070101010a02b603cf
ack
DP vorlauftemperatur succes
boiler - vorlauftemperatur: 95.0
READ 410500010113021c
ack
RCV 41070101011302b603d8
ack
DP Speichertemperatur succes
boiler - Speichertemperatur: 95.0
READ 4105000120000228
ack
RCV 41070101200002dc0007
ack
DP RaumtemperaturSoll succes
boiler - RaumtemperaturSoll: 22.0
READ 4105000120010229
ack
RCV 41070101200102b400e0
ack
DP RedRaumtemperaturSoll succes
boiler - RedRaumtemperaturSoll: 18.0
READ 410500012022024a
ack
RCV 41070101202202d2001f
ack
DP PartyRaumtemperaturSoll succes
boiler - PartyRaumtemperaturSoll: 21.0
READ 410500012006022e
ack
RCV 41070101200602000031
ack
DP HeizkennlinieNiveau succes
boiler - HeizkennlinieNiveau: 0.0
READ 410500012007022f
ack
RCV 41070101200702050037
ack
DP HeizkennlinieSteigung succes
boiler - HeizkennlinieSteigung: 0.5
READ 410500011680019d
ack
RCV 410601011680012fce
ack
DP JAZgesamt succes
boiler - JAZgesamt: 4.7
READ 410500011681019e
ack
RCV 4106010116810130d0
ack
DP JAZHeizen succes
boiler - JAZHeizen: 4.8
READ 410500011682019f
ack
RCV 410601011682012acb
ack
DP JAZWW succes
boiler - JAZWW: 4.2
READ 41050001169001ad
ack
RCV 4106010116900100af
ack
DP COPHeizen succes
boiler - COPHeizen: 0.0
READ 41050001169101ae
ack
RCV 4106010116910100b0
ack
DP COPWW succes
boiler - COPWW: 0.0
READ 410500010500020d
ack
READ boilertemp
READ outsidetemp
READ vorlauftemperatur
READ Speichertemperatur
READ RaumtemperaturSoll
READ RedRaumtemperaturSoll
READ PartyRaumtemperaturSoll
READ HeizkennlinieNiveau
READ HeizkennlinieSteigung
READ JAZgesamt
READ JAZHeizen
READ JAZWW
READ COPHeizen
READ COPWW
READ EinschaltungenVerdichter
nack, timeout
READ boilertemp
READ outsidetemp
READ vorlauftemperatur
READ Speichertemperatur
READ RaumtemperaturSoll
READ RedRaumtemperaturSoll
READ PartyRaumtemperaturSoll
READ HeizkennlinieNiveau
READ HeizkennlinieSteigung
READ JAZgesamt
READ JAZHeizen
READ JAZWW
READ COPHeizen
READ COPWW
READ EinschaltungenVerdichter
READ boilertemp
READ outsidetemp
READ vorlauftemperatur
READ Speichertemperatur
READ RaumtemperaturSoll
READ RedRaumtemperaturSoll
READ PartyRaumtemperaturSoll
READ HeizkennlinieNiveau
READ HeizkennlinieSteigung
READ JAZgesamt
READ JAZHeizen
READ JAZWW
READ COPHeizen
READ COPWW
READ EinschaltungenVerdichter
READ boilertemp
READ outsidetemp
READ vorlauftemperatur
READ Speichertemperatur
READ RaumtemperaturSoll
READ RedRaumtemperaturSoll
READ PartyRaumtemperaturSoll
READ HeizkennlinieNiveau
READ HeizkennlinieSteigung
READ JAZgesamt
READ JAZHeizen
READ JAZWW
READ COPHeizen
READ COPWW
READ EinschaltungenVerdichter
READ boilertemp
READ outsidetemp
READ vorlauftemperatur
READ Speichertemperatur
READ RaumtemperaturSoll
READ RedRaumtemperaturSoll
READ PartyRaumtemperaturSoll
READ HeizkennlinieNiveau
READ HeizkennlinieSteigung
READ JAZgesamt
READ JAZHeizen
READ JAZWW
READ COPHeizen
READ COPWW
READ EinschaltungenVerdichter

```

bertmelis, 2018-04-11:
thanks for testing. Let's continue the NACK discussion in #29

bertmelis, 2018-04-16:
Closing after #34 
Feel free to reopen.

s0170071, 2018-05-16:
So sorry to bother again, but I found some more divide-me-by-10 values. 
Its the energy consumption. The weekly values are stored in COUNTS DPs , the annual vaules in COUNTL. CONTS I can replace with the new COP datatype  but COUNTL still needs a matching /10 counterpart.

Maybe the the /10 DPs should be called COUNTSby10 or so because they are no COP in this case...

bertmelis, 2018-05-16:
Let's think about a generic naming scheme:
Len_Factor_ReturnType like in the code itself?
So you could have a type
`2_Div10_F`--> 2 bytes, division by 10 returning a float
`1_Div10_F` --> 1 byte, division by 10 returning a float
`4_NoConv_UL` --> 4 bytes, no factor returning an unsigned long (uint32_t)
...

Another option would be to just expand the list and add aliases where possible (TEMPS == MODE, COP == E_WEEK or something)

Do you have other idea's?

s0170071, 2018-05-16:
I would propose to add a scaling parameter or a pointer to conversion function to the datapoint. 
As for the naming... if it would stick to C convention it would be more self explainatory. An example: 

COUNTS -> uint16_DP
COUNTL -> uint32_DP
TEMP -> float_DP
STAT -> uint8_DP or even bool_DP
HOURS ->uint16_DP with factor 1/3600
COP -> uint_16_DP with factor 0.1

in the code that would look like this: 
```
VitoWifi.addDatapoint("RaumtemperaturSoll ","boiler",  0x2000,  float_DP,  1);
VitoWifi.addDatapoint("COPWW",              "boiler",  0x1691,  uint16_DP, 0.1);
```
Think of the factor as belonging to the address. Just like the data type does.

bertmelis, 2018-06-26:
I was a bit busy lately, but back in business! I also did some thinking while I was working on another project.

Expect some updates in the very near future.

s0170071, 2018-07-18:
Will try asap. Thanks.