bertmelis/VitoWiFi #33: Optolink connection failed

JueFri, 2018-04-05:
### Installation specifics
* **Heating type:** Vitodens 300-W Typ WB3C with a Vitotronic 200
* **Protocol:** P300
* **Board:** ESP32 mit Adafruit Huzzah32 Feather auf dem Proto Feather Wing 
* **Hardware:** Which hardware implementation did you use?

### Symptom
Optolink connection failed
Could you please recommend a testing strategy for me ? 
What is the best way to check why the communication via the Optolink is not possible ?
Is there anything to do on the Vitotronic to allow the communication via the Optolink ?

@bertmelis 

First of all i would like to say thank you for your excellent work and for your willingness to share this work.

I have a Vitodens 300-W Typ WB3C with a Vitotronic 200 installed and i try to build a Optolink interface based on the building instruction
[ESP32 mit Adafruit Huzzah32 Feather auf dem Proto Feather Wing ](https://github.com/openv/openv/wiki/Bauanleitung-ESP32-Adafruit-Feather-Huzzah32-and-Proto-Wing) and your software [VitoWifi](https://github.com/bertmelis/VitoWifi)

The goal is to use your esp-boiler software with VitoWifi, EspMqtt and WIfiPrinter to integrate the boiler information into openHAB.

Currently i am able to bring the Huzzah32 into the Wlan and connect him with the MQTT broker but i am not able to establish a connection between the Huzzah32 and the Vitotronic via the Optolink.

Currently i use only your **rawOptolink** module to reduce the complexity to establish a connection via the Optolink.

I added some debugging information into the code so i can see that the function **_initHandler** is performed in a loop and that no **stream** seems to be available.

On the Proto Wing board i use a SFH309FA and TSHA4401 for the Optolink. I checked the TSHA4401 with a smartphone camera and can see that she is blinking.

Could you please recommend a testing strategy for me ? 
What is the best way to check why the communication via the Optolink is not possible ?
Is there anything to do on the Vitotronic to allow the communication via the Optolink ?

```cpp
r/*

This example defines 3 datapoints of type "TEMP" in a struct array.
Every minute, the array is iterated.

For each Datapoint, the read value is printed as HEX in Serial1

*/

#include <OptolinkP300.h>

HardwareSerial Serial1(1);
// Choose two free pins
#define SERIAL1_RXPIN 16
#define SERIAL1_TXPIN 17

OptolinkP300 myOptolink;
uint32_t lastMillis = 0;
bool getValues = false;
uint8_t currentIndex = 0;

// Define struct to hold Viessmann datapoints
struct Datapoint {
  uint16_t address;
  uint8_t length;
};
const Datapoint Datapoints[] = {
    0x5525, 2,  // outside temp
    0x0810, 2,  // boiler temp
    0x0812, 2   // DHW temp
};
uint8_t numberOfDatapoints = sizeof(Datapoints) / sizeof(*Datapoints);

//******************************************************************************
// Setup
//******************************************************************************
//
void setup() {
  delay(10000);					// time to reattach serial debug tool
  Serial.println(F("setup - started"));

  Serial.begin(115200);

  // Start Viessmann communication on Serial1
  myOptolink.begin(&Serial1, SERIAL1_RXPIN, SERIAL1_TXPIN);
  Serial.println(F("setup - myOptolink.begin"));

  myOptolink.setLogger(&Serial);
  Serial.println(F("setup - myOptolink.setLogger"));

  Serial.println(F("setup - ended"));
}

//******************************************************************************
// Loop
//******************************************************************************
//
void loop() {
  Serial.println(F("loop - started"));

  // put loop() function inside main loop
  myOptolink.loop();

  if (millis() - lastMillis > 1 * 60 * 1000UL) {  // get values every 1 minute
    lastMillis = millis();
    getValues = true;
    currentIndex = 0;
  }

  if (getValues) {
	  Serial.println(F("loop - getValues"));
    if (!myOptolink.isBusy()) {
      myOptolink.readFromDP(Datapoints[currentIndex].address, Datapoints[currentIndex].length);
    }
  }

  if (myOptolink.available() > 0) {
	  Serial.println(F("loop - available > 0"));
    uint8_t value[4] = {0};
    myOptolink.read(value);
    Serial.print(F("Address "));
    Serial.print(Datapoints[currentIndex].address, HEX);
    Serial.print(F(" has value "));
    for (uint8_t i = 0; i <= 4; ++i) {
      Serial.print(value[i], HEX);
    }
    Serial.println();
    ++currentIndex;
  }

  if (myOptolink.available() < 0) {
	  Serial.println(F("loop - available < 0"));
    Serial.println(myOptolink.readError());
    ++currentIndex;
  }

  if (currentIndex >= numberOfDatapoints) getValues = false;

  Serial.println(F("loop - ended"));
}

```

Any help would be appreciated.

Best regards
 Jürgen

bertmelis, 2018-04-05:
Optolin itself is rather raw and should only be used for diagnostic reasons or when implementing on a device with much less capabilities then the esp8266 or esp32.

I'm working on an example to test which protocol is understood and get the device ID.

But I'm on a tight schedule for the coming period so I'm unable to give very much support at the moment.

Did you try to use the KW protocol?

JueFri, 2018-04-06:
Thank you very much for your quick response.
Until now i only tryed the P300 protocol. I will try to use the KW protocol.

Best regards
 Jürgen

JueFri, 2018-04-06:
This was a good recommendation. With KW as The protocol i got a Connection to the Boiler and the First datapoints arrived.

Thank you very much.

Best regards 
Jürgen

closed 2018-04-07 by JueFri