bertmelis/VitoWiFi #36: Exception29 when interface is disconnected

s0170071, 2018-04-27:
it was all running smoothly until I disconnected the cable and forgot to re-connect it. After a while it crashed.  Looks like a simple queue overflow to me ;-)
I had 60 datapoints queried an an interval of 10s, so the queue was filling up pretty quick. 
I  suggest to limit the queue size to the amount of added data points.

```

Exception (29):
epc1=0x40203e29 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000000c depc=0x00000000

ctx: cont 
sp: 3fff02e0 end: 3fff0560 offset: 01a0

>>>stack>>>
3fff0480:  3ffef4e0 000002d6 3fff04ae 40205c31  
3fff0490:  00000002 00000001 3ffe9288 402066bc  
3fff04a0:  00000000 00000001 3ffe9288 402066bc  
3fff04b0:  3fff0510 0000002a 3ffe8d58 3ffef52c  
3fff04c0:  3fff20f8 0000002b 3ffef4e4 40205b1d  
3fff04d0:  3ffe9287 3fff1e2c 3ffef4e4 40205b1d  
3fff04e0:  3ffe8d2e 3ffef088 3ffef4e4 40205b48  
3fff04f0:  00000000 4bc6a7f0 3ffef4e4 3ffef52c  
3fff0500:  3fff20f8 3fff1e44 3ffef07c 40203f57  
3fff0510:  3fff1e44 00000000 00000000 6353f7ce  
3fff0520:  00000000 3ffef07c 3fff20ac 40203f99  
3fff0530:  3fffdad0 3ffef078 3ffef07c 40202ec0  
3fff0540:  3fffdad0 00000000 3ffef524 40206190  
3fff0550:  feefeffe feefeffe 3ffef540 40100739  
<<<stack<<<

last failed alloc call: 40203F00(504)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

Exception decoder says: 

```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 12 results
0x40203e29: construct  at /home/john/Arduino/scetchbooks/libraries/VitoWifi/VitoWifi.cpp line 274
:  (inlined by) std::deque   >::push_back(VitoWifiBase::Action const&) at /home/john/Arduino/github/Core/Arduino/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_deque.h line 1395
0x40205c31: Print::printNumber(unsigned long, unsigned char) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 92
0x402066bc: Print::write(unsigned char const*, unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 92
0x402066bc: Print::write(unsigned char const*, unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 92
0x40205b1d: Print::write(char const*) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 92
0x40205b1d: Print::write(char const*) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 92
0x40205b48: Print::println() at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 92
0x40203f57: VitoWifiBase::_readDatapoint(Datapoint*) at /home/john/Arduino/scetchbooks/libraries/VitoWifi/VitoWifi.cpp line 274
0x40203f99: VitoWifiBase::readAll() at /home/john/Arduino/scetchbooks/libraries/VitoWifi/VitoWifi.cpp line 274
0x40202ec0: loop at /home/john/Arduino/scetchbooks/reaDatatpoint/readDatapoint/readDatapoint.ino line 175
0x40206190: loop_wrapper at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 58
0x40100739: cont_wrapper at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 81

```

bertmelis, 2018-04-27:
queue size limitation is indeed a to do. I was thinking of twice the number of datapoints to leave room for writing while a readall command is started.

bertmelis, 2018-05-05:
should be solved in V1 branch

closed 2018-06-26 by bertmelis