bertmelis/VitoWiFi #37: Datatypes request

s0170071, 2018-04-27:
### a) device type
2 bytes. Currently using COUNTS as the only 2-byte datatype available:
```

READ 4105000100f80200
ack
RCV 4107010100f80220496c
ack
DP Gerätekennung succes
boiler-Gerätekennung:18720

```
result is 18720 / 0x4920
should be: 0x2049, at least it is referred to as 0x2049 in the table in https://openv.wikispaces.com/Adressen. Not sure who swapped the bytes ;-) If they did- ignore this request.

### b) operating time
currently using COUNT, returning seconds. Hmm... not much use. Would prefer hours as used by the display of the Vitotronic.

bertmelis, 2018-05-03:
I made an update in the datapoint_refactoring branch.
The device type can be queried using the RAW datapoint (remember to set the length to 2).
HOURS datatype was added.
I didn't update the docs yet! And keep in mind that this is not fully tested and there are **breaking changes** in the interface!

s0170071, 2018-05-06:
just wanted to test but I see you made some changes. 
new dp types are 
  
  dpManager.addDP("Raw", "DPs", 0x00, RAW).setLength(4); 
  dpManager.addDP("TempL", "DPs", 0x00, TEMPL); 
  dpManager.addDP("TempS", "DPs", 0x00, TEMPS); 
  dpManager.addDP("Stat", "DPs", 0x00, STAT); 
  dpManager.addDP("CountL", "DPs", 0x00, COUNTL); 
  dpManager.addDP("CountS", "DPs", 0x00, COUNTS); 
  dpManager.addDP("Mode", "DPs", 0x00, MODE); 
  dpManager.addDP("Hours", "DPs", 0x00, HOURS); 
  dpManager.addDP("CoP", "DPs", 0x00, COP); 

Can you quickly cross-reference the new names to the old ones ? I use TEMP now, this has become now TEMPS or TEMPL ?
Is HOURS DP the seconds to hours conversion for the operating time?

bertmelis, 2018-05-06:
temp is now templ. Hours is indeed seconds to hours.

I didn't update the docs (or actually rewrite) yet. Current implemented DPs are listed in datapoint.h

bertmelis, 2018-05-06:
In datapoint.cpp (near the end) you can see which DP belongs to which "conversion". The template parameter stands for "conversion - number of bytes - factor - return type". The return type is chosen as most appropriate for the value it must hold eg: a counter is an unsigned int, an on/off status is a boolean...

bertmelis, 2018-05-07:
I didn't do the math, but I assume memory is indeed an issue.

A removeDatapoint would also imply to scan the queue so there's nothing scheduled using that datapoint. But both datapoint deletion and action deletion are not optimized as they are stored in a vector/queue and they are not intended for random deletions. I would like the lib's use case to be mainly long term connection. A datapoint detective is more of a one-time thing.

So my approach here would be to use the optolink directly and query the addresses one by one, with different length. Just have 1 datapoint of eacht type to be able to do the conversions. A little bit like the rawOptolink and the Datapoints examples combined.

I'll write such an example, but I've got some deadlines to meet at work (so it'll be for the end of the week). If you come up with something, feel free to share and make a pull request!

bertmelis, 2018-05-08:
A question: how do you see the sniffer? I was thinking the following:
Query a address for 1, 2 and 4 bytes; print the possible values and they move on to the next address?

s0170071, 2018-05-08:
Well, almost - yes. 
It is my understanding that the vito does answer ok even if you request 4 bytes from an 4 byte address. It does return an error if you read 2 byte from an 4-byte address. Is this intended ? Seems odd  to me.
So, best would be to try 1, 2, and 4 bytes.

I just tried to read 2/4 bytes from 0x00f8 and 0x0500  . The serial log shows:
```
READ 4105000100f80200
ack
RCV 4107010100f80220496c
ack
DP Gerätekennung succes
boiler-Gerätekennung:18720
READ 4105000100f80402
ack
RCV 4109010100f8042049020b7d
ack
DP Gerätekennung succes
boiler-Gerätekennung:184699168
READ 410500010500040f
ack
RCV 410901010500046d1f0000a0
ack
DP Anzahl_Einschaltungen_Verdichter  succes
boiler-Anzahl_Einschaltungen_Verdichter :8045
READ 410500010500020d
ack
nack, timeout
DP Anzahl_Einschaltungen_Verdichter  error: 1

``` 

If there is no answer or an error- don't print anything. The vast majority of scan attempts will fail.
Time is no factor- memory is. That scan is done once, it may as well take all night. That being said... is it possible that there are data types larger than 4? 

Well and if the output is accidentally formatted in a way that would allow me to copy and paste it into my code with only a few search and replace commands ;-)
E.g. 
DP: 0x1000 MostLikelyType: HOURS; Values: U8 xx U16 xxxx U32 xxxxxxxx I8...

That would allow me to replace:
DP: with VitoWifi.addDatapoint("name ","boiler",
MostLikelyType: with ",",
; Values:  with "); //"

bertmelis, 2018-05-08:
Datapoints are 1, 2, 4 or 8 bytes I think. But I didn't implement the 8 yet (time in BCD for examples I think).

Now I can't write code that evaluates whether a returned value makes sense or not. What I can do is if there is a value, print all possible conversions (currently implemented) but tthen you still have to go through all of them manually.

s0170071, 2018-05-08:
> Now I can't write code that evaluates whether a returned value makes sense or not. What I can do is if there is a value, print all possible conversions (currently implemented) but tthen you still have to go through all of them manually.

Good enough. 
Although an initial rule is simple. Usually (!)
a) Floats are -100< x <100. 
b) Integer should fit comfortably, i.e. are smaller than half the max range.
c) If MSB is set- its an signed int if b) still applies. 
d) if a 16 and a 32 bit value both meet b) I would pick the value that less occupies the max range (i.e. 10% compared to 12%)

winnieXY, 2020-11-11:
Any news on that? Is it possible to read 8bytes for switching times /systime?

bertmelis, 2020-11-11:
I'm still alive. However, I moved temporarily and didn't use this lib for quite some time now. I'll check the code (and branches) and come back to you.

bertmelis, 2023-11-29:
Solved in version 3 of the library. Any length can be queried and raw data is returned.

Will be "formally" published in the coming weeks.

closed 2023-11-29 by bertmelis