bertmelis/VitoWiFi #41: Wrong logging output

s0170071, 2018-08-15:
based on your raw example, I set the data points struct to be: 

```
const Datapoint Datapoints[] = {
    0x0101, 4,  // outside temp
    0x00F8, 2,  // HW Kennung
    0x0113, 4   // solar storage temp
};

```

Reading then returns: 

```
READ 410500010101040c
ack
RCV 410901010101040a01e300ff
ack
Address 101 has value A1E30FE

READ 4105000100f80200
ack
RCV 4107010100f80220496c
ack
Address F8 has value 204900FE

READ 410500010113041e
ack
RCV 41090101011304b603bc029a
ack
Address 113 has value B63BC2FE

```
I spotted two issues:
1. The value 0a01e300 is output as A1E30FE. It omits leading zeros of the single bytes and appends FE.

2. The vaule 2049 is output as 204900FE even tough there were only two bytes data received. Appends FE

3. The value b603bc02 is output as B63BC2FE. Same as 1.

s0170071, 2018-08-15:
maybe myOptolink.available() should return the number of bytes sent by the vito. This way you can allocate a buffer that fits and know how much bytes to process afterwards.

bertmelis, 2023-11-29:
although the issue is real, I probably won't fix this but spend my time on optimizing v3.

closed 2023-11-29 by bertmelis