bertmelis/VitoWiFi #42: need loop() delay of 100ms

s0170071, 2018-08-15:
I programmed a webserver which returns the data point. Work in progress, but seems to work, however, if I loop() too fast I get a nack error.

Here is the cut down version of my website handler.

```
getValues = true;
while (myOptolink.isBusy());  // wait for optolink to become available
myOptolink.readFromDP( address, readLen);  // issue read command
 
    
 while (getValues){
  delay(100);  // need 100ms delay or it will nack
  myOptolink.loop();

  if (myOptolink.available() > 0) {
     getValues=false;
  }
  if (myOptolink.available() < 0) {
     getValues=false;
  }
}
```
please find the delay command. It needs to be at least 100ms for the function to work properly. If it is smaller, it will get a nack error. Does loop() require any special attention when getting called ?

This is the full version of the website handler:
```
void readDP() { //Handler 
String message = "";
String argument=""; 
unsigned int address, readLen;

if (server.args() != 2) { server.send(200, "text/plain", "You must specify exaclty 2 arguments"); return;}
if (server.arg("DP") =="") { server.send(200, "text/plain", "DP argument not found"); return;}
if (server.arg("ReadLen") =="") { server.send(200, "text/plain", "ReadLen argument not found"); return;}
argument=server.arg("DP"); address = strtol(argument.c_str(),0,16); 
argument=server.arg("ReadLen"); readLen = argument.toInt();  
 
getValues = true;
while (myOptolink.isBusy());  // wait for optolink to become available
myOptolink.readFromDP( address, readLen);  // issue read command
 
    
 while (getValues){
  delay(100);  // need 100ms delay or it will nack
  myOptolink.loop();

  if (myOptolink.available() > 0) {
    uint8_t value[40] = {0};
    myOptolink.read(value);
    Serial1.print(F("Address "));
    Serial1.print( address, HEX);
    Serial1.print(F(" has value "));
    for (uint8_t i = 0; i < readLen; ++i) {
      Serial1.print(value[i], HEX);
    }
    Serial1.println();
    for (uint8_t i = 0; i <readLen; ++i) {
          if ( value[i] < 0x10) message +="0";
          message += String( value[i],HEX);
         }
    message += "\n";   
    
    getValues=false;
  }
  if (myOptolink.available() < 0) {
    Serial1.println(myOptolink.readError());
    getValues=false;
  }
}

```

s0170071, 2018-08-15:
I split it up into two functions, now it is working without delay. 
Not sure what is so different now, but since it works now, I close this...
```

 unsigned int readDP (word address, byte readLen){
  unsigned long now = millis(); 
  unsigned int retval = 0xffffffff;

  while (myOptolink.isBusy() && ((millis()-now) < 1000)) delay(10);  // wait for optolink to become available
     if (myOptolink.isBusy() ) return retval;
  myOptolink.readFromDP( address, readLen);  // issue read command
  while (((millis()-now) < 1000)){
        myOptolink.loop();
        if (myOptolink.available() > 0) {
          retval=0;
          uint8_t value[10] = {0};
          myOptolink.read(value);
          Serial1.print(F("Address "));
          Serial1.print( address, HEX);
          Serial1.print(F(" has value "));
          for (uint8_t i = 0; i < readLen; ++i) {
            if ( value[i] < 0x10) Serial1.print("0");
            Serial1.print(value[i], HEX);
            retval = retval <<8;
            retval += value[i];
          }
          Serial1.println();
          Serial1.println(retval,HEX);
          return retval; 
        }
        if (myOptolink.available() < 0) {
          Serial1.println(myOptolink.readError());
          return retval; 
        }
  }
  return retval;
}

void webPage_read() { //Handler 
    String message = "";
    String argument=""; 
    unsigned int address, readLen;
    
    if (server.args() != 2) { server.send(200, "text/plain", "You must specify exaclty 2 arguments"); return;}
    if (server.arg("DP") =="") { server.send(200, "text/plain", "DP argument not found"); return;}
    if (server.arg("ReadLen") =="") { server.send(200, "text/plain", "ReadLen argument not found"); return;}
    argument=server.arg("DP"); address = strtol(argument.c_str(),0,16); 
    argument=server.arg("ReadLen"); readLen = argument.toInt();  
    message = readDP(address,readLen);
    message += "\n";  
    server.send(200, "text/plain", message);//Response to the HTTP request

} 

```

closed 2018-08-15 by s0170071

bertmelis, 2018-08-16:
Are you using the P300 protocol or KW?

s0170071, 2018-08-16:
P300