bertmelis/VitoWiFi #45: Enhancement to Wifi-commands

andreasloe, 2018-08-19:
You call the (great) project VitoWiFi. Does WiFi stand for wireless? In this case I was wondering whether you included some Wifi-Code so that Vito data can be read and written over the internet. Do you have any code to do this?

(Or is this something that someone, whoever that is ;-) , has to do?)

bertmelis, 2018-08-19:
It does stand for WiFi. It's because it sounds ok and the project itself is for esp8266 or esp32 modules which make it easy to connect anything via WiFi.

I didn't include the WiFi code itself because it would conflict with the many frameworks and protocols that serve that purpose.

I personally use standard WiFi code and MQTT from which I can share the code if you want.

andreasloe, 2018-08-19:
MQTT would be great (I am using the optolink together with FHEM and "its" MQTT-surrounding).

Werner-G, 2018-08-21:
This is also exactly my background. Looking forward seeing your ino.file with WIFI and MQTT, Bert!

bertmelis, 2018-08-21:
Remark: I'm using VSCode and Platformio on Windows10.
See https://code.visualstudio.com and http://docs.platformio.org/en/latest/ide/vscode.html

If you're a novice, it may look a bit overwhelming compared to Arduino IDE. But managing dependencies is soo much easier. It also automatically downloads (and upgrades) the complete toolchain to build your firmware.

Should you stick to Arduino IDE you'll have to install VitoWiFi and [AsyncMqttClient](https://github.com/marvinroger/async-mqtt-client) yourself.

**DISCLAIMER**  
I didn't test this code.

platformio.ini
```INI
[platformio]
env_default = esp8266_size

[common_env_data]
platform = espressif8266
board = d1_mini
framework = arduino
lib_deps_external =
    AsyncMqttClient
    VitoWiFi
monitor_speed = 74880 ; same as bootloader

[env:esp8266_size]
platform = ${common_env_data.platform}
board = ${common_env_data.board}
framework = ${common_env_data.framework}
lib_deps = 
    ${common_env_data.lib_deps_external}
build_flags = -Wl,-Tesp8266.flash.4m1m.ld -Wall
monitor_speed = ${common_env_data.monitor_speed}

[env:esp8266_speed]
platform = ${common_env_data.platform}
board = ${common_env_data.board}
framework = ${common_env_data.framework}
lib_deps = 
    ${common_env_data.lib_deps_external}
build_unflags = -Os
build_flags = -Wl,-Tesp8266.flash.4m1m.ld -Wall -O2
monitor_speed = ${common_env_data.monitor_speed}
```
main.cpp
```C++
/*
    copyright 2016 Bert Melis
    License: MIT
*/

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <Ticker.h>
#include <VitoWiFi.h>
#include <AsyncMqttClient.h>

static const char SSID[] = "xxxx";
static const char PASS[] = "xxxx";
static const IPAddress BROKER(10, 0, 100, 2);
static const uint16_t PORT =  1883;
volatile bool updateVitoWiFi = false;

Ticker timer;

VitoWiFi_setProtocol(P300);  // this also initializes VitoWiFi.
DPTemp outsideTemp("outsideTemp", "boiler", 0x5525);
DPTemp boilerTemp("boilertemp", "boiler", 0x0810);
DPTemp dhwTemp("dhwTemp", "hotwater", 0x0812);
DPTemp dhwSolarTemp("dhwSolarTemp", "hotwater", 0x6566);
DPTemp dhwSolarCollTemp("tdhwSolarCollTemp", "hotwater", 0x6564);
DPStat solarPumpStat("solarPumpStat", "hotwater", 0x6552);
DPCountS solarPumpHours("solarPumpHours", "hotwater", 0x6568);

void globalCallbackHandler(const IDatapoint& dp, DPValue value) {
  // nothing to do, all DPs have a custom handler
}

AsyncMqttClient mqttClient;
WiFiEventHandler wifiConnectHandler;
WiFiEventHandler wifiDisconnectHandler;

void connectToWiFi() {
  WiFi.begin(SSID, PASS);
}

void connectToMqtt() {
  mqttClient.connect();
}

void onWifiConnect(const WiFiEventStationModeGotIP& event) {
  timer.once(2, connectToMqtt);
}

void onMqttConnect(bool sessionPresent) {
  mqttClient.publish("HEATING/$status/online", 1, true, "1");
  timer.attach(60, [](){
    updateVitoWiFi = true;
  });
}

void onMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
  if (WiFi.isConnected()) {
    timer.once(2, connectToMqtt);
  }
}

void onWifiDisconnect(const WiFiEventStationModeDisconnected& event) {
  timer.once(2, connectToWiFi);
}

void setup() {
  WiFi.hostname("HEATING");

  VitoWiFi.setGlobalCallback(globalCallbackHandler);
  VitoWiFi.disableLogger();
  VitoWiFi.setup(&Serial);
  outsideTemp.setCallback([](const IDatapoint& dp, DPValue value) {
    char str[9];
    value.getString(str, sizeof(str));
    mqttClient.publish("HEATING/outsideTemp", 1, true, str);
  });
  boilerTemp.setCallback([](const IDatapoint& dp, DPValue value) {
    char str[9];
    value.getString(str, sizeof(str));
    mqttClient.publish("HEATING/boilerTemp", 1, true, str);
  });
  dhwTemp.setCallback([](const IDatapoint& dp, DPValue value) {
    char str[9];
    value.getString(str, sizeof(str));
    mqttClient.publish("HEATING/dhwTemp", 1, true, str);
  });
  dhwSolarTemp.setCallback([](const IDatapoint& dp, DPValue value) {
    char str[9];
    value.getString(str, sizeof(str));
    mqttClient.publish("HEATING/dhwSolarTemp", 1, true, str);
  });
  dhwSolarCollTemp.setCallback([](const IDatapoint& dp, DPValue value) {
    char str[9];
    value.getString(str, sizeof(str));
    mqttClient.publish("HEATING/dhwSolarCollTemp", 1, true, str);
  });
  solarPumpStat.setCallback([](const IDatapoint& dp, DPValue value) {
    mqttClient.publish("HEATING/solarPumpStat", 1, true, (value.getBool()) ? "on" : "off");
  });
  solarPumpHours.setCallback([](const IDatapoint& dp, DPValue value) {
    char str[9];
    value.getString(str, sizeof(str));
    mqttClient.publish("HEATING/solarPumpHours", 1, true, str);
  });

  wifiConnectHandler = WiFi.onStationModeGotIP(onWifiConnect);
  mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  wifiDisconnectHandler = WiFi.onStationModeDisconnected(onWifiDisconnect);
  mqttClient.setServer(BROKER, PORT);
  mqttClient.setClientId("esp-heater");
  mqttClient.setKeepAlive(5);
  mqttClient.setCleanSession(true);
  mqttClient.setWill("HEATING/$status/online", 1, true, "0");
  connectToWiFi();
}

void loop() {
  VitoWiFi.loop();
  if (updateVitoWiFi && mqttClient.connected()) {
    updateVitoWiFi = false;
    VitoWiFi.readAll();
  }
}

```

bertmelis, 2018-08-21:
For more help, head over to https://gitter.im/VitoWiFi/VitoWiFi

Werner-G, 2018-08-21:
Thanks a lot! I move to https://gitter.im/VitoWiFi/VitoWiFi

Werner-G, 2018-08-27:
Now with new Hardware. But my MQTT-spy shows payload of "%.1f" one single time for each topic.

bertmelis, 2018-08-27:
that's not right. Must be a bug in the library.

Werner-G, 2018-08-27:
Which lib?

bertmelis, 2018-08-27:
`value.getString(str, sizeof(str));` prints value into str using format %.1f: floating point with 1 digit after the decimal.
I don't know why it doesn't print.

Does the datapoints.ino example work for you? You don't have to be connected to the optolink when trying that example.

bertmelis, 2018-09-14:
closing, added link to fully working example.

closed 2018-09-14 by bertmelis

Bojkas1985, 2019-11-13:
Hello bertmelis,

i am new with platformio. I install Visual Studio and extension Platformio.
I using NodeMCU v3 and planed build this optolink: https://github.com/openv/openv/wiki/Bauanleitung-ESP8266

What i must do? Only create these two files platformio.ini and main.cpp and edit adress for WO1C.

That is all?

Thank you
Ivica