bertmelis/VitoWiFi #48: protocol blocked after error

JPAPLOUZ, 2018-10-17:
Hello
I have experienced your vitoWifi library on ESP8266nodeMCU and a Vitobens 222W and the protocol P300 works well
Nevertheless if a mistake happens on the connection the system doesn't recover.

if it is a bad identifier or bad length problem, vitoWifi produces after only reading with "optolink.available <0" even on good adresses and lengths requests.

if it is a transmission problem like the optical path obscured, the system doesn't respond anymore (probably due to  an UART crash)
Can you help to fix  these 2 problems 

Best regards 

...

Be as complete as you can!

### Extra info
Please add relevant debug information like debug messages, (decoded) stack traces...

*You don't have to provide all the info, but less is not always more!*

bertmelis, 2018-10-19:
I'll look into this.

bertmelis, 2018-10-29:
is it possible to retry with the _setState and _setAction blocks uncommented? You can find the code at the bottom of OptolinkKW.hpp. Afterwards, please share the output so we can look where it halts.

bertmelis, 2023-11-29:
Stale issue, closing.

closed 2023-11-29 by bertmelis