bertmelis/VitoWiFi #49: interface with Open Energy Monitor

Pukkahq, 2018-10-18:
### Installation specifics
* **Heating type:**  Vitodens 200 i think
* **Protocol:** unknown
* **Board:** nothing yet
* **Hardware:** 

Viessmann 222f boiler.

We have the above unit and I understand the hardware side of an esp. we have a home energy monitoring system [https://openenergymonitor.org/](url) i am wondering if it would be possible for your code to be modified to output to emoncms, I have very little programing skills but a lot of enthusiasm.

[https://github.com/emoncms/emoncms](url) Main emoncms info
[https://github.com/emoncms/emoncms/blob/master/Modules/input/Views/input_api.php](url) Input API

They already use esp8266 in some of there products.
[https://github.com/OpenEVSE/ESP8266_WiFi_v2.x](url)

bertmelis, 2018-10-19:
I don't know how emoncms works, but I could possibly make an example to interface with that system. i'm not sure however that TLS is going to work on a esp8266. As the interface is a HTTP-API, I'll have to use a third party lib (https://github.com/boblemaire/asyncHTTPrequest).

Pukkahq, 2018-10-19:
Hi, Thanks for your help
I run a local emoncms so wont need TLS, 
They also have a very nice interface and way of attaching the ESP8266 to a wifi network. 
our emonpi will aslo allow you to publish to a MQTT feed.

bertmelis, 2018-10-19:
what do you mean with "attaching an esp8266 to a network"?

Pukkahq, 2018-10-19:
sorry I mean the process they go through to connect the eps8266 to your wifi network

https://github.com/openevse/ESP8266_WiFi_v2.x/#user-guide

bertmelis, 2023-11-29:
Closing, integration is up to the user. I'm happy to help where possible and adjust the library if needed.

closed 2023-11-29 by bertmelis