bertmelis/VitoWiFi #52: Callback with old value after write

maihacke, 2018-11-07:
I create a bridge between vitowifi and openhab via mqtt.
Therefore I implemented a method which publishes all read values via mqtt as global callback.
Therefore I call readAll values in the esp loop. Then the global callback of vitowifi publishes all values via mqtt. That works quite well.
For writing thr sketch listens to mqtt and call writeDataPoint accordingly.
The problem is, that immediately after the write the global callback is called with the old value and this outdated value gets published via mqtt.

Is there a reason why vitowifi calls the callback with the old value after a write? I think the callback should not be invoked at all for writes.

What do you think?

maihacke, 2018-11-07:
Im using KW. I will double check what happens and report it here.

maihacke, 2018-11-08:
It was just an error in my type conversion logic.
Btw: Thanks for this cool project.

closed 2018-11-08 by maihacke

maihacke, 2018-11-09:
Code is here: https://github.com/maihacke/vitohab
Please notice it is heavily in progress.
Writing is not ready yet...

FulvioSpelta, 2018-11-09:
Thank you very much. I'd like to have a red-only device at the moment.

maihacke, 2018-11-09:
I will write a readme for openhab item definitions etc when I find the time...

bertmelis, 2018-11-09:
For the record: there's an example in [issue 45](https://github.com/bertmelis/VitoWiFi/issues/45).

I'm running a service on my RPI3 (Raspbian latest) with the code in this repo: [Mqtt2InfluxDB](https://github.com/bertmelis/Mqtt2InfluxDB)