bertmelis/VitoWiFi #53: KW runs into Timeout

Empor-co, 2018-11-15:
### Installation specifics
* **Protocol:** KW
* **Board:** ESP8266 as well as ESP32, both adafruit feather

### Symptom
Values are not read correctly. It sometimes workes nicely but degrades suddenly. I can see Readall being called but the handles not involved. If values are returned they are mostly bogous like temperature being 128.5 C. After uncommenting in the header file the debugging I do see the following:

`READ outsideTemp
READ boilertemp
READ pump
TestOptolink action: 1
Optolink state: 2
Optolink state: 3
Optolink state: 4
READ 5525
timeout
Optolink state: 0
Optolink action: 3
Optolink action: 0
DP outsideTemp error: 1
Optolink action: 1
Optolink state: 1
Optolink state: 2
Optolink state: 3
Optolink state: 4
READ 0810
timeout
Optolink state: 0
Optolink action: 3
Optolink action: 0
DP boilertemp error: 1
Optolink action: 1
Optolink state: 1
Optolink state: 2
Optolink state: 3
Optolink state: 4
READ 2906
timeout
Optolink state: 0
Optolink action: 3
Optolink action: 0
DP pump error: 1
Optolink state: 1
 `
When I blindly "fire" some requests directly to the serial I do get the right values returned.

Any ideas?

Thank you in advance,

empor

bertmelis, 2018-11-15:
I'll look into this.

How much time is there between the messages? Less then 1 second? About one second or about 5 seconds?

Empor-co, 2018-11-15:
Less than 1 second, more or less a continous stream of messages per datapoint.

Empor-co, 2018-11-18:
It might be connected to some UART problems of the ESP32:
https://www.esp32.com/viewtopic.php?t=4960&p=21544#p21544

bertmelis, 2023-11-29:
Stale issue, feel free to reopen of the issue is still relevant.

closed 2023-11-29 by bertmelis