bertmelis/VitoWiFi #54: Platformio build not possible [.pioenvs\esp8266\firmware.elf] Error 1

Wuifi, 2019-02-11:
Software Tools used:
Windows 10
Atom 1.25.1
Platformio IDE Home 2.0.0·Core 3.6.4
-----------------
I am not able to build the software. I tried the installed Atom / Platformio as well as a portable version.
The Error is with both build environments (ESP8266 and ESP32) the same:
```
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\esp8266\firmware.elf] Error 1
```
[error log.txt](https://github.com/bertmelis/VitoWiFi/files/2853121/error.log.txt)

**What I did so far:**
I took the example from the [esp8266-heating](https://github.com/bertmelis/esp8266-heating) and copied the files platformio.ini and main.cpp into the relevant folders adapted everything as decribed and tried the build.
I have no idea how to proceed and debug further on. Guess it's a very common problem many of you faced with.
Can anybody provide a stable set to build the software with the VitoWifi-master repository? That would be the best way. Or at least point your finger into the direction I might find the problem.

[image]

Many thanks in advance!

 

For information: * **Hardware:** Which hardware implementation did you use?
### Installation specifics
* **Heating type:** Vitotronic 200 KW1
* **Protocol:** KW 
* **Board:** ESP8266 or ESP32 (others?), possible brand/make (NodeMCU, Wemos D1 mini...)

bertmelis, 2019-02-12:
Could you share the complete error message?

If you're using platformio, you could just download the example. All other libraries will be downloaded automatically.

You're on Linux/Mac/Windows?

Wuifi, 2019-02-12:
Hello,

It's all in already. see the first lines of the issue for my platform setup. It's Win10, Atom and Platformio.
The (unfortunately partial) Error Log is included in the text-file! Once I get the platform running again with the portable versions, I will add the complete one again - or close the issue in case I figured it out on my own.
 
Regards
Wolfgang

Wuifi, 2019-02-12:
Ok,
it`s still a Windows PC, I forgot ;-)! 
With a clean out of the box portable version of Atom / platformio it ran immediately.
[image](https://user-images.githubusercontent.com/45028259/52667633-a7f03e80-2f11-11e9-86c9-d04a0afebce6.png)

closed 2019-02-12 by Wuifi

bertmelis, 2019-02-12:
For the record, on Windows I have better experiences with VSCode and platformio then Atom.

Don't know if there's a portable version though.