bertmelis/VitoWiFi #61: ESP 8266 with Loxone SmartHome

katzendrama, 2019-07-28:
Hi Bert, I want to control my Viessmann Vitodens 200 with Loxone. But there is no ready hardware and software. I have done many projects on NodeMCUs oder Arduinos. Is the ESP8266 library directly flashable? I want to receive UDP Messages from the NodeMCU with all the readings and I want to send UDP Messages to the NodeMCU with the new values to set. Is there any code for this application or how can I realise this?

Greets from Germany, Vincent.

bertmelis, 2019-07-28:
There's no out of the box solution I'm afraid.
This library isn't directly flashable but needs to je integrated in you own firmware.

I also don't know the least about Loxone. Don't they have a TCP interfaces instead of an UDP? UDP messages could get lost...

Anyways, creating firmware that reeds your heater and sends the value(s) via UDP takes some time but is not that hard.

katzendrama, 2019-07-28:
I can program Arduinos... but I currently dont understand your project as there is no API description...

bertmelis, 2019-07-29:
You want to work with V1 or V2 (wip)?
Do you need the P200 of the KW protocol?
Do you have a list of datapoints you want to control?
Do you have the needed hardware and does it work?

katzendrama, 2019-07-29:
I have a NodeMCU Lolin V3 and IR LEDs and all kinds of resistors and a IR Receiver.

katzendrama, 2019-07-29:
And my Viessmann device is the Vitodens 200 (of 2013) with 19kW...

katzendrama, 2019-08-05:
Hi Bert, my Adapter now works with KW protocol and your Blynk example.

Ok, so I want to control these Datapoints:
The Forerun Temperature
Only Hotwater / Hotwater and Heating
Set Roomtemperature
Set Waterboiler Temperature

And I want to get these values:
The actual Forerun temperature
Which state... Only Warmwater / Both Warmwater Heating
the set roomtemperature
and the set boilertemperature

bertmelis, 2019-08-05:
- Did you use the code from "master" or from "V2"?
- what type is your Viessmann device? (and what type of control you have?)

Datapoints are defined by address, length and type. I don't know all of them, let alone for all the different Viessmann devices.

katzendrama, 2019-08-05:
I used the code from master

I have a Viessmann Heating Vitodens 200-W with 19kW and the DIY Adapter for ESP8266 with the photodiode and the IR LED and a NodeMCU V3

katzendrama, 2019-08-05:
Ok... now it works very well... So I need all datapoints and the bytes for it you know. And for every of them if they are writable...

Thanks...

bertmelis, 2019-08-06:
Have a look in the "docs" folder from this repo. There are some files with datapoints that work for me (2013 Vitodens 200 with Vitotronic 200).

bertmelis, 2023-11-29:
I believe this issue is solved?
Feel free to reopen if not.

closed 2023-11-29 by bertmelis