bertmelis/VitoWiFi #62: Implement GWG protocol

Arty68, 2020-01-02:
Hello I have a vitodens 100 wb1. 
I have the "Bauanleitung-ESP8266" + esp-link and vitocontold (I also tried with python script which are also working for low level read).
I just began research for the adress.
I had some results with this : https://github.com/openv/openv/issues/442
I am very interrested about your work.
Would be nice to use vitowifi for GWG protocol also

bertmelis, 2020-01-03:
I don't own a Viessmann heating anymore (at least temporary). I can help but I can't test obviously.

Your code is in your repo?

Arty68, 2020-01-03:
Hi Bertmelis
Thank you for answering so quickly !
First of all Happy New Year :)
Yes I have put it [here](https://github.com/Arty68/VitoWiFi/tree/master/src) the moment.
I am not sur if everything is ok.
You can have a look when you have time
[image]

Arty68, 2020-01-23:
Hello

I have continued the GWG but not have tested it.

https://github.com/bertmelis/VitoWiFi/compare/master...Arty68:master

bertmelis, 2020-01-28:
Sorry to keep you waiting. I'll have a look in the next few days.

Now, I see you have it implemented using the master branch. My idea would be to -as soon as it is working- move it over to the V2 branch. I'm happy to do this for you.

Arty68, 2020-01-28:
No Problem. I will try to test it once I get a little free time.
Thank you to take time to review it.

Arty68, 2020-04-07:
Hello

I have continued the GWG implementing but still not have tested it.
It's not compiling yet . I still have problem with the template yet in the VitoWifi.cpp line 126.
Also have a GWG test in the example directory.

bertmelis, 2020-04-15:
I'm still alive :smile:
I'll take a look. It's OK for you to test it against the V2 branch? I'd like to move away from the current master branch. V2 should be easier to deal with.

I can also make an example for you. What do you want to do with the obtained data? Display on webserver? Send via MQTT?

Arty68, 2020-04-15:
Thanks for the answer. I will try to use the v2 branche. I am not a C++ expert but I will try. Glad that you are still alive :) . Take care. I am from France by the way. Alsace to be more precise.

alfredolo72, 2023-11-29:
hello, any news on gwg protocol ?

bertmelis, 2023-11-30:
first version (untested!) added in v3-development

alfredolo72, 2023-11-30:
i will be happy to test it. I have a Vitodens 200. Right now cannot compile it for esp32, i try to fix errors

bertmelis, 2023-12-01:
Ours is older and is able to use the VS2 protocol. I suggest you also use this.

bertmelis, 2023-12-08:
Closing, GWG is implemented in v3. It's untested though. Create a new issue when problems arise.

closed 2023-12-08 by bertmelis