bertmelis/VitoWiFi #63: [OT] WifiPrinter

sude22, 2020-01-05:
Hi Bert,
sorry for this OffTopic issue, but I could not find out how to contact you direct.
Could you please give a hint what happened to the "https://github.com/bertmelis/WifiPrinter" library?

I used it in some former projects....

bertmelis, 2020-01-05:
It was hopelessly outdated and I remover the public repo. I can look on my computer to see if I have something left there...

But otherwise, it is not that hard to recreate from scratch. You just want to print messages to clients connected to a specific port?

sude22, 2020-01-06:
I have a local copy, but I was just curious if it vanished for some special reason.
It was very useful for debugging when the serial connection was used for other stuff (like the Vito communication ;)
I removed it in most of the newer projects because it seems that it leads to Wifi connection errors with the homie framework, but I didn't had a closer look at that.

Thanks for your good work.

bertmelis, 2020-01-06:
If I remember correctly it uses the blocking version of the TCP server so it could very well mess up if mixed with Homie (which is async).

sude22, 2020-01-06:
OK, that would be an explanation. I thought it was async implemented because it used a loop method...

bertmelis, 2020-01-28:
Drafted something quickly: https://github.com/bertmelis/WiFiPrinter

closed 2020-01-28 by bertmelis