bertmelis/VitoWiFi #64: Circuit Example

jnsw, 2020-03-09:
Following the pictures of the real world build, it seems that RX is connected with 3.3V over that 10k ohm resistor but in this diagram it seems the transistor has three legs 

https://github.com/bertmelis/VitoWiFi/blob/master/doc/circuit.png

what is correct now?

bertmelis, 2020-03-09:
The transistor only has 2 legs. The circuit is wrong. It is triggered by light so it has no gate. It's on the to do list but this project is on the bottom of the pole because I don't have a Viessmann heater for the moment.

nouknouk, 2020-07-04:
Hi,

you've got example circuits schemas in the GitHub openv Wiki.
Examples: 
- [esp8266](https://github.com/openv/openv/wiki/Bauanleitung-ESP8266)

[image]

- [esp32 (the same)](https://github.com/openv/openv/wiki/Bauanleitung-ESP32-Adafruit-Feather-Huzzah32-and-Proto-Wing)

[image]

On my side, I wired a circuit for my ESP32 based on those examples, composed only of IR LEDs and resistors ; result was not stable enough and lead to many issues.

I switched to the circuit given [in example for the RaspBerry Pi ](https://github.com/openv/openv/wiki/Bauanleitung-RaspberryPi), with additionals transistors. It is fully compatible with ESP32 microcontroller (given youuse GPIO 16 and 17 for RX and TX) and despite being just a bit more complex (2 transistors) it works much, much better.

[image]

bertmelis, 2020-07-04:
Which transistor/diode you're using? If I recall correctly, the suggested types in the ESP circuit are obsolete. So maybe the RPI circuit is more tolerant for the newer parts.

nouknouk, 2020-07-04:
Hi,

I use the same diodes as those recommended at the origin: SFH487-2 and SFH209FA. The SFH487-2 is discontinued but but still can find some on ebay.

For transistors, the same as those on the Rpi circuit too: 2N3906 for emission and the BC547B (PNP one) for reception.

My circuit is now functionning like a charm.

[image]

Btw, thanks for your work ; even if I plan to develop my own 'driver' for my heater/home automation server through USB/serial, your code give me a good basis for furtherinspiration.

bertmelis, 2023-11-29:
Circuit corrected in v3

closed 2023-11-29 by bertmelis