bertmelis/VitoWiFi #66: Please tell me the request address V100 boiler pump in operation or not / It is not in the list of addresses of Viessmann-adresses

dima-357, 2020-11-21:
### Installation specifics
* **Heating type:** eg. 20CB - Vitodens 200 with Vitotronic 200
* **Protocol:** KW or P300
* **Board:** ESP8266 or ESP32 (others?), possible brand/make (NodeMCU, Wemos D1 mini...)
* **Hardware:** Which hardware implementation did you use?

### Symptom
What happens? (crash, unresponsive, nothing at all...)  
When does it happen? (after several hours, immideately...)  
What would you expect?  
...

Be as complete as you can!

### Extra info
Please add relevant debug information like debug messages, (decoded) stack traces...

*You don't have to provide all the info, but less is not always more!*

dima-357, 2020-11-21:
Please tell me the request address V100
boiler pump in operation or not /
It is not in the list of addresses of Viessmann-adresses

bertmelis, 2020-11-22:
If it's not in the list, then I don't know.

I don't have a list of every data point for every type of heating.

closed 2020-11-22 by bertmelis