bertmelis/VitoWiFi #67: V1: protocol blocked after invalid DP address

s0170071, 2021-06-23:
### Installation specifics
ESP8266 + Vitotronic200 + Vitocal300

### Symptom
What happens? (crash, unresponsive, nothing at all...)  
I cannot communicate with the heater when I use a wrong DP address. It does not recover.
As I am currently scanning the heater for all responsive data points, this cannot be avoided by nature.

When does it happen? (after several hours, immideately...)  
immediately

What would you expect?  
It should recover when I use a correct address, instead I need to reboot. 

I have a workaround. In Optolink300.cpp, change line

`    _setState(RESET);`

to 

` _setState(IDLE);`

I have to admit I did not dig too deep into the protocol stack and I cannot explain whats wrong with reset, but it did solve my issue. 
@Bert, do you know why RESET is inappropriate at this point ? Technically a wrong DP address is not an error.

bertmelis, 2021-06-24:
I'll have a look, but I'm leaving for holiday and only return in 3 weeks...

bertmelis, 2023-11-29:
Stale issue, feel free to reopen of the issue is still relevant.

closed 2023-11-29 by bertmelis