bertmelis/VitoWiFi #68: No Data with ESP8266, but data with ESP32

Kasper027, 2022-02-05:
### Installation specifics
* **Heating type:** Vitocal 200-s with WO1C
* **Protocol:** KW or P300 both work
* **Board:** ESP8266 NodeMCU and ESP32 NodeMCU

### Symptom
Hey. With the Example Basic-ESP32 and an ESP32-Board I get Data on the Serial-Monitor by ArduinoIDE. 
Connecting it to the ESP8266 NodeMCU on the GPIO1+3 there are no Data on the Serial-Monitor. Only two ? came on the Monitor.  Can someone help me?

bertmelis, 2022-02-16:
Sorry, this issue slipped my mind.

first thing that I think about is hardware. The library isn't different for the two platforms. Could you check the wiring? Did any error messages appear?

Kasper027, 2022-02-19:
Hey @bertmelis,
I have already tried the following:
a Wemos D1 Mini, the reading head on RX and TX,
then only question marks appear over the SerialMonitor in the Arduino IDE.
I tried the same on a NodeMCU V3.4 ESP8266. Also connected to RX and TX. But the same result.
Or do I have to set up the connection differently with the ESP8266? Isn't the USB interface also Serial0?

Do you have a picture of how you connected it?
With the ESP32 board everything runs without problems.

Kasper027, 2022-02-19:
I tested a little more.
I connected an FTDI adapter to D4. There I get basic data (Setup finished...). 
Then I added a Serial1.print("loop") to the loop to see if it works so far. The "loop" is also written out cleanly.
But all other expenditures are not written.
`void loop() {
  static unsigned long lastMillis = 0;
  if (millis() - lastMillis > 60 * 1000UL) {  // read all values every 60 seconds
    Serial1.println("loop");
    lastMillis = millis();
    VitoWiFi.readAll();
  }
  VitoWiFi.loop();
}`

Do you have any other ideas I can test? I have already exchanged RX and TX, but no success either

bertmelis, 2022-02-21:
It's been a while since I used the library myself. (because of circulstances I don't have a viessmann boiler at the moment).

Anyway...

ESP8266 only has 1.5 UARTs onboard. You can use `Serial` for the optolink, but then you only have the .5 (TX only, `Serial1`) for other communication. You might want to use SoftwareSerial for the optolink and keep `Serial` for the serial monitor as it is connected to the USB. I haven't adapted to library to be used with softwareserial though.

Another option of course is not to use the serial monitor but publish the values with MQTT for example.

Kasper027, 2022-02-21:
Hey @bertmelis 
i have tested with the half Serial connection and an FTDI adapter. But there are no Data to print out. Ok, if you don’t have any ideas more, I will change the other Code from Esp8266 to ESP32

bertmelis, 2022-02-21:
Euh, an ftdi adapter doesn't matter. The ESP8266 only has one full UART. So if you use that for the optolink, you cannot properly communicate.

How did you wire the esp?

Kasper027, 2022-02-21:
By the 8266 I connect the optolink to Rx and Tx, 3,3V and Gnd. The Adapter to D4. 
by starting the „Setup finished“ we’re printet to the Adapter. And I have take a `Serial1.print(„loop“)` tobte Loop part will also printet by the adapter. But no Data from the Optolink

Kasper027, 2022-02-21:
[image]
This is my setup

bertmelis, 2022-02-21:
Hmmm, that's strange. Can you enable the logger and set it to Serial1?

Kasper027, 2022-02-21:
Did you mean this?

[image]

bertmelis, 2022-02-21:
Yes it should give you some debug statements.

Kasper027, 2022-02-21:
[image]

can you say me, whats wrong?

bertmelis, 2022-02-21:
Try &Serial1

You forgot the ampersant.

Kasper027, 2022-02-21:
here is the Serial Output
`22:17:57.274 ->                                                               r l⸮⸮r⸮ ⸮#⸮n⸮⸮⸮⸮⸮p⸮<⸮⸮⸮ ⸮x⸮⸮ǒ⸮⸮p⸮nn⸮⸮;⸮nĒ⸮⸮⸮b⸮$⸮rrp⸮n⸮⸮⸮ ⸮⸮⸮⸮l⸮⸮⸮⸮⸮b⸮n⸮⸮n⸮䎐⸮⸮⸮b⸮ľ~⸮n⸮⸮⸮⸮⸮ l`⸮⸮⸮#⸮n⸮⸮;nr⸮⸮⸮;⸮⸮?r⸮;p⸮n⸮⸮܀⸮r⸮⸮⸮b⸮⸮⸮8⸮b⸮n⸮⸮n⸮⸮⸮b⸮ľ~⸮n⸮⸮⸮r⸮⸮#⸮n⸮⸮;nr⸮⸮⸮;⸮⸮l`⸮8⸮⸮⸮;⸮⸮?l`Ğ⸮⸮⸮⸮⸮;⸮⸮n⸮⸮`⸮Setup finished...
22:18:29.345 -> loop
22:18:29.345 -> READ outsideTemp
22:18:29.345 -> READ boilertemp
22:18:59.340 -> loop
22:18:59.340 -> READ outsideTemp
22:18:59.340 -> READ boilertemp
`

bertmelis, 2022-02-22:
Can you share your while sketch?
Do you use the same optolink with the esp32?

Kasper027, 2022-02-22:
i used your example basic.ino
`
/*
This example defines three datapoints.
The first two are TEMPL type datapoints and have their own callback.
When no specific callback is attached to a datapoint, it uses the global callback.
Note the difference in return value between the callbacks:
for tempCallback uses value.getFloat() as TEMPL datapoints return a float.
globalCallback uses value.getString(char*,size_t). This method is independent of the returned type.
*/

#include <VitoWiFi.h>

VitoWiFi_setProtocol(P300);

DPTemp outsideTemp("outsideTemp", "boiler", 0x0101);
DPTemp boilerTemp("boilertemp", "boiler", 0x010D);
//DPStat pumpStat("pump", "heating1", 0x2906);

void tempCallbackHandler(const IDatapoint& dp, DPValue value) {
  float fahrenheit = 0;
  fahrenheit = (5.0 / 9) * (value.getFloat() + 32);
  Serial1.print(dp.getGroup());
  Serial1.print(" - ");
  Serial1.print(dp.getName());
  Serial1.print(": ");
  Serial1.println(fahrenheit, 1);  // print with 1 decimal
}

void globalCallbackHandler(const IDatapoint& dp, DPValue value) {
  Serial1.print(dp.getGroup());
  Serial1.print(" - ");
  Serial1.print(dp.getName());
  Serial1.print(" is ");
  char value_str[15] = {0};
  value.getString(value_str, sizeof(value_str));
  Serial1.println(value_str);
}

void setup() {
  outsideTemp.setCallback(tempCallbackHandler);
  boilerTemp.setCallback(tempCallbackHandler);
  VitoWiFi.setGlobalCallback(globalCallbackHandler);  // this callback will be used for all DPs without specific callback
                                                      // must be set after adding at least 1 datapoint
  Serial1.begin(115200);
 VitoWiFi.setLogger(&Serial1);
 VitoWiFi.enableLogger();
 VitoWiFi.setup(&Serial);
  
  Serial1.println(F("Setup finished..."));
}

void loop() {
  static unsigned long lastMillis = 0;
  if (millis() - lastMillis > 60 * 1000UL) {  // read all values every 60 seconds
    Serial1.println("loop");
    lastMillis = millis();
    VitoWiFi.readAll();
  }
  VitoWiFi.loop();
}
`

bertmelis, 2022-02-22:
To be honest: I don't know why it works with ESP32 and not with ESP8266. On both platforms the HardwareSerial libraries have the same API and hardware is practically the same: both platforms work on 3.3V.  . I also cannot test anything in real life because I don't have a Viessmann heater.

You might want to try the V2.0 branch: https://github.com/bertmelis/VitoWiFi/tree/v2.0.0 but I doubt that's going to change anything.

Kasper027, 2022-02-22:
i will tested it this evening

Kasper027, 2022-02-22:
I have now tested version 2. Same behavior.
I will now run everything on the ESP32 and try this again sometime.

I have another problem, but I'll open a separate topic for that as it has nothing to do with this.

bertmelis, 2023-11-29:
Is this still relevant?
Please consider testing v3 and report again if the issue persists.

closed 2023-12-08 by bertmelis