bertmelis/VitoWiFi #69: Temperature-Datapoint with 3 Bytes

Kasper027, 2022-02-24:
Hey @bertmelis 
my heat pump has some data points with 3 bytes. For example temperatures.
Address 0101 is the outside temperature on 2 bytes.
Address 01C1 is the outside temperature on 3 bytes.
the first two bytes are the value (little endian) and the 3rd is a status.
How can I read this data

Here is the data from the raw data:

20:55:44.021 -> READ 410500010101020a
20:55:44.021 -> ack
20:55:44.058 -> RCV 410701010101023d004a
20:55:44.058 -> ack
20:55:44.058 -> Address 101 has value 3D000F8
20:55:44.058 -> READ 4105000101c103cb
20:55:44.091 -> ack
20:55:44.091 -> RCV 4108010101c1033d00404c
20:55:44.091 -> ack
20:55:44.091 -> Address 1C1 has value 3D0400F8

Unfortunately, the HEX is sometimes put together incorrectly. According to the raw data, this should be 3D0040F8.
So converted 003D-->61-->6.1°C
40-->Status (can be omitted)

But how can I create that as a data point?

bertmelis, 2022-02-24:
Which version of the lib do you use?

Kasper027, 2022-02-24:
I testet it with the Master. Not the Verion 2.

bertmelis, 2022-02-24:
```cpp
// first create a converter class, and inherit from DPType
class conv_temp_3B : public DPType {
 public:
  void encode(uint8_t* out, DPValue in) {  // encode can be left empty when the DP is only used for reading.
    // empty
  }
  DPValue decode(const uint8_t* in) {
    int16_t tmp = in[2] << 8 | in[1];  // keep endianess in mind! input is LSB first
    DPValue out(tmp / 10.0f);
    return out;
  }
  const size_t getLength() const {
    return 3;
  }
};
// the typedef is optional. If you should consider making a PR,
// please do make a typedef for consistency.
typedef Datapoint<conv_temp_3B> DPTemp3B;

// now you can use the newly created type:
DPTemp3B myDatapoint("name", "group", 0x1234);
```

I quickly made something based on the example in the readme. Not tested...

bertmelis, 2023-11-29:
Example code given.

in v3 raw data is available so you can more easily transform/convert data yourself.