bertmelis/VitoWiFi #71: "Betriebsart" as example delivers always  "0"

GnomiBerlin, 2022-08-20:
### Installation specifics
* **Heating type:** Vitotronic  200 KW2
* **Protocol:** KW 
* **Board:** ESP8266 Wemos D1 mini 
* **Hardware:** own IR components

### Symptom
What happens? some values do not work

example:
DPMode getBetriebArtM1("getbetriebartm1","betriebsarten", 0x2301);     //HK1 0=Abschaltb,1=nur WW,2=heiz+WW, 3=DauernRed,3=Dauer Norma.

I also tried address 2500 and 2501, but always a "0" is delivered on the MQTT interface.
Normally now I expect a "1" for only WW

Is the Address wrong or the coding?

Thanks

ondras12345, 2022-08-20:
On my device (`0x00F8` = `0x20CB`, whatever model number that is), `0x2301` also reads `0x00`.
`0x2323` works as "betriebsart", both read and write.

bertmelis, 2022-08-21:
I'm sorry but I can't give support on addresses. Viessmann decided to be selfish and keep the data internal. You can find some documents in the docs/ folder but that's all I've got.

GnomiBerlin, 2022-08-21:
Thanks ondras12345 und bertmelis,

20CB is a viessmann VScotHO1 Vitodens 200, if the tables are o.k. But 0x2323 is not working with my V200KW2.

And if Viessmann means, that this is the right decision in an open IOT world, where all are going to open source-o.k. 
Lots of big SW companies decided different because of these large IOT communities, which are also a product decision for the end customers.

I will find it out with some more effort.

Thanks again for answering

bertmelis, 2022-08-21:
You may find some data in this repo: https://github.com/openv/openv/wiki or maybe on the Home Assistant or Openhab forums.

I temporarily moved and don't have a Viessmann boiler at the moment. Otherwise I could test on mine.

closed 2022-08-21 by bertmelis