bertmelis/VitoWiFi #74: VitoWiFi.readDatapoint() only on MQTT request

GnomiBerlin, 2022-09-04:
Installation specifics
Heating type: Vitotronic 200 KW2
Protocol: KW
Board: ESP8266 Wemos D1 mini
Hardware: own IR components

**Question:**
I want a MQTT update not in a loop but for some values **on request only**. But it works not as expected?!

Example:
if i use below line with all related configurations, I get an MQTT: VITOWIFI/gettimerzirkusa with its value each minute.
`getTimerZirkuSa.setCallback(DP8TimerCallbackHandler);  //fuer Timer`

then I added:
in the MQTT receiving part, for a **new MQTT message** "retrievetimerzirkusa" to get the DP "gettimerzirkusa" on request:
  ```
if (strcmp(topic, "VITOWIFI/retrievetimerzirkusa") == 0) {
    VitoWiFi.readDatapoint(getTimerZirkuSa);
  }
```
and
`mqttClient.subscribe("VITOWIFI/retrievetimerzirkusa", 0);`

to get no updates per minute, I took out:
`//getTimerZirkuSa.setCallback(DP8TimerCallbackHandler);  //fuer Timer`

What did I not understand?

Thanks.

bertmelis, 2022-09-04:
It it possible to share the whole code (remove credentials)?

GnomiBerlin, 2022-09-04:
no nice, but in the middle of test code: wanted not to see the code here so just a time limited link:
https://www.dropbox.com/sh/iz97u42pn6ci3j7/AACdy5TvW8gvLrIYsngqVjoJa?dl=0

these are all the changed files.

GnomiBerlin, 2022-09-11:
I found a way for my problem.

I used the group identifier for "synchronize" and "on_request" datapoints.
e.g.:
```
DPTemp getTempA("gettempa", "synchronize", 0x0800);               //Aussentemperatur
DPStat getAlarmStatus("getalarmstatus", "synchronize", 0x0847);   //Sammelstoerung Ja/Nein  0x0A82 0x7579
DP8Timer getAnlagenzeit("getanlagenzeit", "synchronize", 0x088E);   //Anlagenzeit  DPHours 0x088E DPCount
DP8Timer getTimerZirkuSa("gettimerzirkusa", "on_request", 0x2228);   //Zirkulationspumpe Zeiten Samstag
DP8Timer setTimerZirkuSa("settimerzirkusa", "on_request", 0x2228);   //Zirkulationspumpe Zeiten Samstag
...
```
in the loop part I changed VitoWiFi.readAll() --> VitoWiFi.readGroup("synchronize") for all on minute received values:
```
void loop() {
  if (!bStopVito) {
    VitoWiFi.loop();
    if (updateVitoWiFi && mqttClient.connected()) {
      updateVitoWiFi = false;
      VitoWiFi.readGroup("synchronize");   //VitoWiFi.readAll();
    }
  }
...
```
for all "on_request" datapoints I added MQTT  subscriptions and its handling:
```
if (strcmp(topic, "VITOWIFI/settimerzirkusa") == 0) {
    uint64_t setTimer = strtoull(payload, NULL, 16);
    DPValue value(setTimer);
    VitoWiFi.writeDatapoint(setTimerZirkuSa, value);   // 06000730160018300000000000000000 = FFFFFFFF93803B30
    VitoWiFi.readDatapoint(getTimerZirkuSa);
  }
  if (strcmp(topic, "VITOWIFI/retrievetimerzirkusa") == 0) {
    uint64_t setTimer = strtoull(payload, NULL, 16);
    DPValue value(setTimer);
    VitoWiFi.readDatapoint(getTimerZirkuSa);
  }
```
I hope this is clearly described.

Is the idea o.k. or any other ideas? If yes I can close the issue.

bertmelis, 2022-09-11:
I don't think there is anything wrong with this way of work.

GnomiBerlin, 2022-09-11:
With workaround solved

closed 2022-09-11 by GnomiBerlin