bertmelis/VitoWiFi #75: Serialmonitor "chksum 0x2d"

Trainerr, 2022-09-04:
### Installation specifics
* **Heating type:** Vitoladens 300c with Vitotronic 200 KW6B
* **Protocol:** KW
* **Board:** ESP8266 (Wemos D1 mini.)
* **Hardware:** SFH485P and SFH309FA

### Symptom
i can't see any IR blinking signal with my smartphone camera. 

i got this message after boot up.
```
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld
```
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                            
it looks like my esp8266 didn't boot up correctly or is this message normal?

bertmelis, 2022-09-05:
ESP8266 only has 1 full UART so if you want to use VitoWiFi, you can't use the Serial monitor at the same time. You can use libraries like this on https://github.com/LennartHennigs/ESPTelnet to print messages like telnet.

Trainerr, 2022-09-05:
so how can i check if it is working if i don't get any values via Mqtt?

My IR LED is also not shining.

bertmelis, 2022-09-11:
Instead of printing debug messages to Serial, you print them to another "printer". You could use this library: https://github.com/JoaoLopesF/RemoteDebug

```cpp
#define HOST_NAME "vitowifidebug"
#include <RemoteDebug.h>
RemoteDebug debug;

void setup() {
  // ...
  VitoWiFi.enableLogger();
  VitoWiFi.setLogger(&debug);
}

void loop() {
  // ...
  debug.handle();
}
```

bertmelis, 2023-11-29:
Closing because of configuration error.
Feel free to reopen or create a discussion if help is needed.

closed 2023-11-29 by bertmelis