bertmelis/VitoWiFi #77: Blocking after 6 hours

GnomiBerlin, 2022-09-28:
nstallation specifics
Heating type: Vitotronic 200 KW2
Protocol: KW
Board: ESP8266 Wemos D1 mini
Hardware: own IR components

Issue:
Sometimes the data transfer stopped and I had to reboot the VitoWifi ESP Board.
Then I made a watchdog in NodeRed, that after 3 Minutes of no MQTT data the link xxx.xxx.xxx.xxx/reboot is called by NodeRed to automatically restart the board. This works.

**But** there was a unexpected picture given: the **reboot often happens each 6 hours**. So is there any idea, why the VitoWifi is blocked after exactly 6 hours?

Thanks

bertmelis, 2022-09-28:
So the board is still responding to MQTT, only VitoWiFi stopped working?
That's gonna be a hard nut to crack.

GnomiBerlin, 2022-09-28:
Yes, the .../reboot seems to work, but the 60 seconds synchronize get... results are not sent anymore.
I also have no real idea at the moment.

bertmelis, 2023-11-29:
Issue still valid?

If so, please consider testing v3.

closed 2023-12-08 by bertmelis