bertmelis/VitoWiFi #78: Errors on  high MQTT request rate

GnomiBerlin, 2022-10-09:
Installation specifics
Heating type: Vitotronic 200 KW2
Protocol: KW
Board: ESP8266 Wemos D1 mini
Hardware: own IR components

Issue: errors on  high MQTT request rate

I used the new common developed [;-)] conv8_1_Timer  for the setting of the weekday on/off timing of heating, water and circulation pump.
If I set the 4 periods for a single week day all is working (one 8 Byte command)
If I send 5 commands (Monday-Friday) from NodeRed with MQTT directly to VitoWifi, only Monday is correct and the others contain "FFFFFFFFFFFFFFFF". This is also seen in the Debug WRITE output from VitoWifi.
If I put 1 second delay in NodeRed between each weekday setting command, all is working.

I expected a queued handling of the set commands without errors. 

Any idea?

bertmelis, 2022-10-09:
Which mqtt client you're using?

GnomiBerlin, 2022-10-09:
Mosquitto

bertmelis, 2022-10-09:
> Mosquitto

on the esp

GnomiBerlin, 2022-10-09:
No, on a Raspberry Pi 4 with enough load capacity.

bertmelis, 2022-10-09:
How does your esp communicate? And which library do you use for communication?

GnomiBerlin, 2022-10-10:
First I did not fully understand your question but maybe this is the point where I have to ask for the errors? #77,#78

I started with:  https://github.com/Schnup89/OpenV_NodeMCU

**Arduino IDE:**
Using board 'd1_mini' from platform ...esp8266\3.0.2
Using core 'esp8266' from platform .... esp8266\3.0.2

Bibliothek ESP8266WiFi in Version 1.0 
Bibliothek Ticker in Version 1.0 
Bibliothek VitoWiFi-master in Version 1.0.2 i
Bibliothek async-mqtt-client-develop in Version 0.9.0 
Bibliothek ESPAsyncTCP-master in Version 1.2.2 
Bibliothek ESP8266WebServer in Version 1.0 
Bibliothek ESP8266mDNS in Version 1.2
Bibliothek ESP8266HTTPUpdateServer in Version 1.0 
Bibliothek RemoteDebug-2.1.1 in Version 2.1.1

bertmelis, 2022-10-10:
I don't have a Viessmann boiler at the moment so I can't test myself. You didn't give me any code to look into. Yes, the error may be in VitoWiFi but it can also be somewhere else.  You also didn't give me any debug output. I don't have a crystal ball.

bertmelis, 2023-11-29:
Issue still valid?

If so, please consider testing v3.

closed 2023-12-08 by bertmelis