bertmelis/VitoWiFi #79: Keine Verbindung mit ESP32 über USB-Host möglich

DominikWild, 2022-10-23:
### Installation specifics
* **Heating type:** eg. 20CB - Vitodens 300-T with Vitotronic 200 Typ KW1
* **Protocol:** KW
* **Board:** ESP32 NODEMCU von AZ-Delivery
* **Hardware:** USB Adapter V4 mit  CH340G (neue Version) + Hobbytronics USB Host Controller Board V2.4
### Symptom
What happens? Ich bekomme keine Verbindung, bzw. die LED am Optolink-Adapter bleibt aus.

...

### Extra info
Ich habe meinen ESP32 über die PINS 21 und 22 mit dem USB Host Controller Board V2.4 von Hobbytronics verbunden.
An diesem Board ist der Optolink-Adapter angeschlossen. Mit dem Testprogramm bleibt die LED am Adapter aus und ich bekomme keine Daten.

`/*

This example defines three datapoints.
The first two are DPTemp type datapoints and have their own callback.
When no specific callback is attached to a datapoint, it uses the global callback.

Note the difference in return value between the callbacks:
for tempCallback uses value.getFloat() as DPTemp datapoints return a float.
globalCallback uses value.getString(char*,size_t). This method is independent of the returned type.

*/

```
#include <VitoWiFi.h>

VitoWiFi_setProtocol(KW);

DPTemp outsideTemp("outsideTemp", "boiler", 0x5525);
DPTemp boilerTemp("boilertemp", "boiler", 0x0810);
DPStat pumpStat("pump", "heating1", 0x2906);

void tempCallbackHandler(const IDatapoint& dp, DPValue value) {
  float fahrenheit = 0;
  fahrenheit = (5.0 / 9) * (value.getFloat() + 32);
  Serial.print(dp.getGroup());
  Serial.print(" - ");
  Serial.print(dp.getName());
  Serial.print(": ");
  Serial.println(fahrenheit, 1);  // print with 1 decimal
}

void globalCallbackHandler(const IDatapoint& dp, DPValue value) {
  Serial.print(dp.getGroup());
  Serial.print(" - ");
  Serial.print(dp.getName());
  Serial.print(" is ");
  char value_str[15] = {0};
  value.getString(value_str, sizeof(value_str));
  Serial.println(value_str);
}

void setup() {
  outsideTemp.setCallback(tempCallbackHandler);
  boilerTemp.setCallback(tempCallbackHandler);
  VitoWiFi.setGlobalCallback(globalCallbackHandler);  // this callback will be used for all DPs without specific callback
                                                      // must be set after adding at least 1 datapoint
  VitoWiFi.setup(&Serial2, 21, 22);  // 
  Serial.begin(115200);
  Serial.println(F("Setup finished..."));
  VitoWiFi.setLogger(&Serial);
  VitoWiFi.enableLogger();
}

void loop() {
  static unsigned long lastMillis = 0;
  if (millis() - lastMillis > 20 * 1000UL) {  // read all values every 60 seconds
    lastMillis = millis();
    VitoWiFi.readAll();
  }
  VitoWiFi.loop();
}`
```

Schreibe ich in einem anderen Programm Daten an die serielle Schnittstelle (PIN 21 + 22) leuchtet die LED.
Schließe ich den Adapter direkt an einen PC mit Windows an, bekomme ich eine Verbindung zur Heizung.

Hat evtl. jemand eine Idee, warum es mit der VitoWiFi nicht funktioniert?

Der USB Host erkennt den angeschlossenen Optolink-Adapter und alle Einstellungen stehen auf Standard .

[image]

bertmelis, 2022-10-23:
I don't understand your hardware setup.
what does the usb host controller do? Doesn't your ESP32 nodemcu have an onboard usb converter?
The optolink is is most setups directly connected to the UART pins. how did you do it?

bertmelis, 2023-11-29:
Is this issue still valid?

bertmelis, 2023-12-01:
I'm going to close this issue. I have no experience whatsoever with the usb host board.

One of the features I am going to add is an USB host mode. This should be possible for the ESP32-S2 and -S3 boards. No ETA though.

closed 2023-12-01 by bertmelis