bertmelis/VitoWiFi #80: Wemos D1 mini / Serial-Pins

lsrs7, 2022-10-25:
### Installation specifics
* **Heating type: Vitodens 200
* **Protocol: P300
* **Board: ESP8266 Wemos D1 mini

Using self-made optolink with softwareserial and looks good. It doesn't work with RX/TX (1/3 ports), but with D7/D6.
VitoWifi for 8266 is not providing possibility to change RX/TX-Pin instead of ESP32 code. 

Any hint to run VitoWifi with customized Pins?

Cheers,
Lutz

gitzila, 2022-11-18:
@lsrs7 Hi, maybe this helps. I tried to use VitoWifi with D1 Mini an changed the file "OptolinkP300.cpp":

Change line 52+ to the following:

void OptolinkP300::begin(HardwareSerial* serial) {
  serial->begin(4800, SERIAL_8E2);
  delay(1000);
  serial->swap();
  delay(1000);
  _stream = serial;
  // serial->flush();
}

The RX/TX is then swapped to Pins 13 an 15. I had no time to complete the project, but testing with some leds shows communications seems to use the Pins 13/15 now...

empty88, 2023-02-05:
@lsrs7 
Hi, I changed OptolinkP300.cpp and OptolinkKW.cpp and made VitoWifi SoftwareSerial compatible. Then I used SoftwareSerial with other pins instead of RX / TX.
Check [my project](https://github.com/empty88/OptoLink)

Best regards

Bojkas1985, 2023-02-05:
@lsrs7 Hi, will this also work with the Vitocal 242s heat pump? Anyway, I flashed your code using VSC and platformia, just changed the port. In the available Wifi I found optolink_config, which I connected to both from computer and mobile, but I can't connect to ip 192.168.0.1, it refuses the connection "ERR_CONNECTION_REFUSED".
Any advice? I have tried both Wemos D1 mini and NodeMCU, only the LED lights up but nothing happens.

Thank you

Translated with www.DeepL.com/Translator (free version)

empty88, 2023-02-05:
@Bojkas1985 
Hi,
to be honest I have no idea why you get the connection refused error. I have never had issues like this.
Maybe you can check if the serial logging output gives a hint.

Best regards

Bojkas1985, 2023-02-07:
@empty88 Hi, I finally solved it by opening it in edge. The problem is that after entering the WLAN and the mqtt broker, I cannot connect to the administration. When I turn it off and on, it won't connect to the network again. the last message from the terminal is start webserver. In that administration, I can then set the codes under which I want Wemos to communicate with Viessmann, or do I have to do it in the code?

empty88, 2023-02-07:
@Bojkas1985 hm, strange behavior...
maybe the WLAN connection infos are not correct, So it tries to connect about 200 times and then starts the optilink_config AP again, the complete serial log output would be interesting, e.g. which IP address it is assigned to? Did it start in configmode or not?

The codes have to be set in the code.

bertmelis, 2023-11-29:
Is this issue still valid?

closed 2023-12-08 by bertmelis