bertmelis/VitoWiFi #81: Example for DPRaw

Sirmike2, 2022-10-29:
### Installation specifics
* **Heating type: Vitodens 200 
* **Protocol:** P300
* **Board:** ESP8266

### Symptom
get DPRaw not work

Hello,

I implemented your program accordingly on my Viessmann heating. Works really great.
But I'm not sure which version of the heater and control I really have. I wanted to read out the address 0x00F8. 
Unfortunately I'm not that good at programming. I wanted to use DPRaw to get the hex value. But I can't get it implemented. 
It should work somehow with .setLength() and getRaw(), but I don't get it.

I implemented the other points in such a way:

`
  DPMode getBetriebArtM1("getbetriebartm1","betriebsarten", 0x2301);

  void tempSCallbackHandler(const IDatapoint& dp, DPValue value) {
  int nValue = value.getU8();
  char outName[30] = "VITOWIFI2/";
  strcat(outName,dp.getName());
  mqttClient.publish(outName, 1, true, String(nValue).c_str()); 
}

\\Void Setup

getBetriebArtM1.setCallback(tempSCallbackHandler);
`

That works ok, but how should I do this with DPRaw ?
Can you include an example or just post it here? Would be really great.

bertmelis, 2022-11-02:
Didn't test the code. DPRaw works the same as the other datapoint types. It just returns the unformatted byte stream instead of the human readable value.

```cpp
DPRaw raw_dp("Type", "DPs", 0x00F8);

void typeCallback(const IDatapoint& dp, DPValue value) {
  size_t length = value.getRawLength();
  uint8_t* type = new uint8_t[length];
  value.getRaw(type);
  // type is now in 'type'

  // you can also print:
  char buf[100];  // 100 chars should be enough
  value.getSting(buf, 100);
  Serial.println(buf);
}
```

bertmelis, 2023-11-29:
Feel free to reopen if help is needed.

Otherwise, check v3 of this library. access to raw data is available by design.

closed 2023-11-29 by bertmelis