bertmelis/VitoWiFi #82: Can you explain a bit more how to install the library in Arduino IDE 2.0?

LFrank2021, 2022-11-22:
I just installed Arduino IDE 2.0 and assume I need to install the VitoWifi library.
Can you maybe elaborate a bit how this can be achieved?
I assume I need a .zip file?
Or is it as simple as downloading the src tree and zipping it?

Thank you in advance.
Kind regards

bertmelis, 2022-11-22:
This repo has a great explanation: https://github.com/gnea/grbl/wiki/Compiling-Grbl#via-the-arduino-ide-all-platforms-recommended-for-all-users.

LFrank2021, 2022-11-22:
Thanks, that helped.
I am getting a lot of error messages:
`PATH/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\libraries\VitoWiFi\DPTypes.cpp.o: in function `_ZN7convRaw6encodeEPh7DPValue':
PATH\Documents\Arduino\libraries\VitoWiFi\src/DPTypes.cpp:28: multiple definition of `_ZN7convRaw6encodeEPh7DPValue'; PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\sketch\DPTypes.cpp.o:PATH\Documents\Arduino\sketch_nov22a/DPTypes.cpp:28: first defined here
PATH/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\libraries\VitoWiFi\DPTypes.cpp.o: in function `_ZN7convRaw6decodeEPKh':
PATH\Documents\Arduino\libraries\VitoWiFi\src/DPTypes.cpp:33: multiple definition of `_ZN7convRaw6decodeEPKh'; PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\sketch\DPTypes.cpp.o:PATH\Documents\Arduino\sketch_nov22a/DPTypes.cpp:33: first defined here
PATH/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\libraries\VitoWiFi\DPTypes.cpp.o: in function `_ZN10conv2_10_F6encodeEPh7DPValue':
PATH\Documents\Arduino\libraries\VitoWiFi\src/DPTypes.cpp:40: multiple definition of `_ZN10conv2_10_F6encodeEPh7DPValue'; PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\sketch\DPTypes.cpp.o:PATH\Documents\Arduino\sketch_nov22a/DPTypes.cpp:40: first defined here
PATH/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\libraries\VitoWiFi\DPTypes.cpp.o: in function `_ZN10conv2_10_F6decodeEPKh':
PATH\Documents\Arduino\libraries\VitoWiFi\src/DPTypes.cpp:45: multiple definition of `_ZN10conv2_10_F6decodeEPKh'; PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\sketch\DPTypes.cpp.o:PATH\Documents\Arduino\sketch_nov22a/DPTypes.cpp:45: first defined here
PATH/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\libraries\VitoWiFi\DPTypes.cpp.o: in function `_ZN7DPValue5getU8Ev':
PATH\Documents\Arduino\libraries\VitoWiFi\src/DPValue.hpp:90: multiple definition of `_ZN10conv1_1_US6encodeEPh7DPValue'; PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\sketch\DPTypes.cpp.o:PATH\Documents\Arduino\sketch_nov22a/DPValue.hpp:90: first defined here
PATH/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\libraries\VitoWiFi\DPTypes.cpp.o: in function `_ZN10conv1_1_US6decodeEPKh':
PATH\Documents\Arduino\libraries\VitoWiFi\src/DPTypes.cpp:56: multiple definition of `_ZN10conv1_1_US6decodeEPKh'; PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\sketch\DPTypes.cpp.o:PATH\Documents\Arduino\sketch_nov22a/DPTypes.cpp:56: first defined here
PATH/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\libraries\VitoWiFi\DPTypes.cpp.o: in function `_ZN7DPValue7getBoolEv':
PATH\Documents\Arduino\libraries\VitoWiFi\src/DPValue.hpp:83: multiple definition of `_ZN9conv1_1_B6encodeEPh7DPValue'; PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\sketch\DPTypes.cpp.o:PATH\Documents\Arduino\sketch_nov22a/DPValue.hpp:83: first defined here
PATH/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\libraries\VitoWiFi\DPTypes.cpp.o: in function `_ZN9conv1_1_B6decodeEPKh':
PATH\Documents\Arduino\libraries\VitoWiFi\src/DPTypes.cpp:64: multiple definition of `_ZN9conv1_1_B6decodeEPKh'; PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\sketch\DPTypes.cpp.o:PATH\Documents\Arduino\sketch_nov22a/DPTypes.cpp:64: first defined here
PATH/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\libraries\VitoWiFi\DPTypes.cpp.o: in function `_ZN7DPValue6getU32Ev':
PATH\Documents\Arduino\libraries\VitoWiFi\src/DPValue.hpp:104: multiple definition of `_ZN10conv4_1_UL6encodeEPh7DPValue'; PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\sketch\DPTypes.cpp.o:PATH\Documents\Arduino\sketch_nov22a/DPValue.hpp:104: first defined here
[... cut ...] \OptolinkP300.cpp.o: in function `_ZN12OptolinkP3009setLoggerEP5Print':
PATH\Documents\Arduino\libraries\VitoWiFi\src/OptolinkP300.cpp:389: multiple definition of `_ZN12OptolinkP3009setLoggerEP5Print'; PATH\AppData\Local\Temp\arduino-sketch-D9C76B2DED5C65FD7ECA31FF96048A99\sketch\OptolinkP300.cpp.o:PATH\Documents\Arduino\sketch_nov22a/OptolinkP300.cpp:389: first defined here
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1`

Any help is appreciated.

LFrank2021, 2022-11-22:
Ignore that, I found the reason.
Failed try to copy the src in same directory as sketch.

closed 2022-11-22 by bertmelis