bertmelis/VitoWiFi #84: VitoWiFi 2.0 Debug funcionality

LFrank2021, 2022-12-19:
Hi there,
I hope you are well and the lack of updates is only because of the non existing test/development environment.

I switched to v2.0 and am missing the ability to debug the VitoWiFi library.
I cannot find anything similar to the _printer object.

Since I had issues with the v1 (aka master) branch and was at least able to trace to it to a timeout situation.
But with v2 without a deeper look into the optolink connection I fear I will not be able to locate errors (hardware or software).

Hope you can at least find the time to send a quick answer (I am alive).
Kind regards
Frank

LFrank2021, 2022-12-23:
I finally get some data.
Using the version 2.0 I had to partially work on KW implementation.
I have no idea if this is really correct.
It started working as soon as I deactivated all 'Serial.print' statements.
Seems indeed on an ESP8266 USB-Serial and Serial for VitoWiFi cannot coexist.
Tried to implement SoftwareSerial for VitoWiFi but failed utterly.

Still have some issues with incorrect sensors and writing is yet something I need to work out.
@bertmelis How did you intend to write data to addresses?
With 2.0 there is no longer a setWrite function, right?
```writeSuccess = vitodens200.write(getTempWWsoll, setTempf);``

bertmelis, 2023-01-17:
SoftwareSerial should work. I remember having it implemented once. But afterwards stopped using it because I wanted to have the values transmitted via WiFi anyway.

But yes, the project is deep down in the freezer because it's of no practical use for me at the moment. Not being able to test in real life is hurting me.

Now, I don't want it to go to waste. I should open it up to collaborators. And maybe redesign: I started this project when my C++ design experience was unexisting.

LFrank2021, 2023-01-17:
Great hearing from you.
I currently own a VT200 and managed to get the D1 mini and your library (v2) to work for the most part.
I am extremely struggling with finding all my datapoints and seem to get a few 0x05 from datapoint queries resulting in high values (1285°).
I managed to get data with the KW protocol.
Could maybe retry using P300 but do not know if my VT supports this (did not seem that way).
Would be interested in the integration of the GWG protocol to see if that works better.

If I can assist in any way let me know. (for example let you peek at my GH repo).

bertmelis, 2023-11-29:
Is this issue still relevant?
If so, please also consider testing v3.

bertmelis, 2023-12-01:
v3 is *completely* (?) different. It also uses standard Arduino logging instead of printing to your defined printer. Actually, logging inside the library is for debug purposes.

But i can accommodate if things are wanted differently.

bertmelis, 2023-12-11:
Logging in v3 is done through the Arduino framework API calls. Preferences wrt output should be managed there.

closed 2023-12-11 by bertmelis