bertmelis/VitoWiFi #86: Solved: Vitocal300 WO1B does not start with VitoWifi attached

s0170071, 2023-04-14:
I attached VitoWifi to the Wo1B control unit. Works great!

One thing that has been annoying though is that  I always needed to power up the heater first. Dangerous low WAF after mains power glitches. 

Seems like the heater  goes into some boot mode when it detects communication right after power up. 
I looked into it and [here](https://github.com/openv/openv/wiki/Protokoll-300) is a nice description of the protocol.

So basically all the ESP needs to do after power up is wait for the 0x5 character to arrive before it fires up vitowifi.
I put this code into my setup() 

`
  unsigned int timeout=millis();
  Serial.begin(4800); 
  while (Serial.available()) { Serial.read();  } // remove whatever is in the queue.
  while (millis()< timeout +20000){          // 20 second timeout.
     delay(1000);   // wait a second
     if (Serial.available() > 0) {  // has a byte arrived ?
      if (  Serial.read() == 0x05)   break;  // skip the rest of the while loop if it is 0x05
    } 
  }
` 

Home this helps someone. 
@bertmelis: nothing to do for you here ;-)

novaktom, 2023-10-06:
@s0170071 I have ESP on battery backup, but not Vitocal unit. In case of power outage ESP does not restart, so setup() would not help. Don't you know how to solve that please? Restart ESP periodically?  Or close (0x04) and initialize communication periodically (every 5 min or so)?

s0170071, 2023-10-07:
The thing is that when the Vitotronic starts, it listens for communication. If it sees any, it goes into some bootloader mode. I did not investigate this further as the solution is to just don't talk to it when it restarts. Seems like your're better off not hooking the ESP up to the backup supply. 
In fact, I hooked mine up to the internal switch of the Vitotronic so that it gets rebooted when I switch on the heater.

I did this because my wife would hate me if the heating fails and cannot be restarted while I am away...

The only alternative that I can think of is that you monitor the heater power supply with your ESP.

bertmelis, 2023-10-08:
I remember suffering this issue too. There's no real bullet-proof solution. If your ESP is on mains power, you could add a delay before VitoWiFi starts listening and hope your heater has booted meanwhile.

PS I recently moved back to a house with a Viessmann heater so I'm planning to support more actively again. I have yet to build a module before I can start though.

bertmelis, 2023-12-08:
Closing. This has to be taken care of by the implementation.

closed 2023-12-08 by bertmelis