bertmelis/VitoWiFi #88: Up-to-date verision on PlatformIO

j0ta29, 2023-10-09:
[Hi @bertmelis,
first of all thanks for the contribution of this great library.
I'm the author of a component on the ESPHome platform that brings Viessmann Optolink support into the world of ESPHome/Home Assistant (see https://github.com/esphome/esphome/pull/4453 and [OpenV Wiki](https://github.com/openv/openv/wiki/ESPHome-Optolink)). My component uses your library and since ESPHome's build process is based on PlatformIO it uses the last released version 1.0.2 of your library in the registry of PlatformIO.
One user asked for support of 3-byte-datapoints (also see https://github.com/bertmelis/VitoWiFi/issues/69). As this request is from last year (which means much younger than version 1.0.2), I'm wondering if you plan to release a more up-to-date version of your library on PlatformIO.
Best Regards,
Jürgen

bertmelis, 2023-10-10:
Long story short: a few years ago, I moved abroad to a house without Viessmann heater. So although I created this library, I didn't have a use case myself anymore. The project went into hibernation.

Not so long ago, I moved back into our old house and I do plan to revive the project. I have a solar system and it's not immediately obvious something's wrong. So I want to monitor that.

First thing on the planning is to build a module again to connect physically to the heater.
Meanwhile I'll create a issue/discussion to ask for user input. A few years have gone by and I feel a bit more comfortable with C++ so I might just create a version 3 and start from there. I saw some updates on the [openv-wiki](https://github.com/openv/openv/wiki/Protokoll-300) which I need to incorporate.

Expect some action soon.

s0170071, 2023-10-11:
Hi @bertmelis ,
meanwhile I created a ESP8266 app that uses this library to push the readings to MQTT. Works better with OpenHab. It uses high priority reads for status messages and low prio for temperatures. 
It also has http access if you want to use that instead of MQTT. Small probing page for DPs too. And some homeassistant autodiscovery that I ended up not using  after all ;-)

You would find it here: https://github.com/s0170071/VitoWifiApp

bertmelis, 2023-11-29:
I'm busy with v3 but in the meantime I've also published v1.1.0 to the platformio registry.
I'm going to skip in v2 for now.

closing the issue. this will solve itself in the near future.

closed 2023-11-29 by bertmelis