bertmelis/VitoWiFi #109: How to use softwareserial?

marcus2208, 2024-09-22:
### Installation specifics
* **Heating type:** Vitoladens 300-T with Vitotronic 200 KW1
* **Protocol:** KW
* **Board:** ESP8266 (Wemos D1 mini clone)
* **Hardware:** The one in your example, but connected to (RX) GPIO4 and (TX) GPIO5
* **Modification:** Softwareserial for optolink and telnet for output

### Symptom
What happens? Timeout on all datapoints  
When does it happen? Always  

The softwareserial and the hardware connected to is confirmed working, because I can communicate via telnet to Vitotronic, without vitowifi, using a different sketch. I just can´t find out, how to pass the softwareserial to your library, so vitowifi is working with softwareserial. When I try, at least the IR-LED is flashing, so someting is working already. But all datapoint's results are timeouts. My guess is, vitowifi is using the softwareserial as hardwareserial, resulting in SERIAL_8E2 instead of SWSERIAL_8E2. But since I´m not a high-level-skilled C programmer, I might be wrong.

### My code
```
#include <ESP8266WiFi.h>
#include "SoftwareSerial.h"
const int sRX = 4; // software RX on D2(GPIO 4)
const int sTX = 5; // software TX on D1(GPIO 5)
EspSoftwareSerial::UART Optolink(sRX, sTX);

#include <WiFiManager.h>
#include "BaseOTA.h"
#include <VitoWiFi.h>

#define MAX_SRV_CLIENTS 1
WiFiManager wifiManager;
WiFiServer server(23);
WiFiClient serverClient;
const char deviceName[] = "Vitotronic";

VitoWiFi::VitoWiFi<VitoWiFi::VS1> vitoWiFi(&Optolink);
bool readValues = false;
uint8_t datapointIndex = 0;
unsigned long lastReceived = 0;

VitoWiFi::Datapoint datapoints[] = {
  VitoWiFi::Datapoint("outsidetemp", 0x5525, 2, VitoWiFi::div10),
  VitoWiFi::Datapoint("boilertemp", 0x0810, 2, VitoWiFi::div10),
  VitoWiFi::Datapoint("watertemp", 0x0812, 2, VitoWiFi::div10),
  VitoWiFi::Datapoint("heatpump", 0x2906, 1, VitoWiFi::noconv),
  VitoWiFi::Datapoint("loadpump", 0x0845, 1, VitoWiFi::noconv),
  VitoWiFi::Datapoint("circpump", 0x0846, 1, VitoWiFi::noconv),
  VitoWiFi::Datapoint("burnerrun", 0x0842, 1, VitoWiFi::noconv),
  VitoWiFi::Datapoint("burnererror", 0x0883, 1, VitoWiFi::noconv)
};

void onResponse(const uint8_t* data, uint8_t length, const VitoWiFi::Datapoint& request) {
  // raw data can be accessed through the 'response' argument
  serverClient.print("Raw data received:");
  for (uint8_t i = 0; i < length; ++i) {
    serverClient.printf(" %02x", data[i]);
  }
  serverClient.print("\n");

  // the raw data can be decoded using the datapoint. Be sure to use the correct type
  serverClient.printf("%s: ", request.name());
  if (request.converter() == VitoWiFi::div10) {
    float value = request.decode(data, length);
    serverClient.printf("%.1f\n", value);
  } else if (request.converter() == VitoWiFi::noconv) {
    // in this example, the response is one byte
    serverClient.printf("%s\n", (data[0] > 0) ? "ON" : "OFF");
  }
}

void onError(VitoWiFi::OptolinkResult error, const VitoWiFi::Datapoint& request) {
  serverClient.printf("Datapoint \"%s\" error: ", request.name());
  if (error == VitoWiFi::OptolinkResult::TIMEOUT) {
    serverClient.print("timeout\n");
  } else if (error == VitoWiFi::OptolinkResult::LENGTH) {
    serverClient.print("length\n");
  } else if (error == VitoWiFi::OptolinkResult::NACK) {
    serverClient.print("nack\n");
  } else if (error == VitoWiFi::OptolinkResult::CRC) {
    serverClient.print("crc\n");
  } else if (error == VitoWiFi::OptolinkResult::ERROR) {
    serverClient.print("error\n");
  }
}

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  delay(5000); //BOOT WAIT

  WiFi.setSleepMode(WIFI_NONE_SLEEP);     // disable WiFi sleep for more performance
  WiFi.setOutputPower(10);                // 0 to 20.5 [dBm] inclusive, 0.25 [dBm] steps
  WiFi.hostname(deviceName);

  wifiManager.setDebugOutput(false);
  wifiManager.autoConnect(deviceName);

  OTAinit();

	server.begin();
	server.setNoDelay(true);

  vitoWiFi.onResponse(onResponse);
  vitoWiFi.onError(onError);
  vitoWiFi.begin();
}

void loop() {
  ArduinoOTA.handle();

	if (server.hasClient()) {
		AcceptConnection();
	}

  if (millis() - lastReceived > 60000UL) {  // read all values every 60 seconds
    lastReceived = millis();
    readValues = true;
    datapointIndex = 0;
  }

  if (readValues) {
    if (vitoWiFi.read(datapoints[datapointIndex])) {
      ++datapointIndex;
    }

    if (datapointIndex >= 8) {
      readValues = false;
    }
  }

  vitoWiFi.loop();
}

void AcceptConnection() {
	if (serverClient && serverClient.connected()) {
		serverClient.stop();
	}

	serverClient = server.accept();
}

```
### The essential parts
```
#include "SoftwareSerial.h"
const int sRX = 4; // software RX on D2(GPIO 4)
const int sTX = 5; // software TX on D1(GPIO 5)
EspSoftwareSerial::UART Optolink(sRX, sTX);
VitoWiFi::VitoWiFi<VitoWiFi::VS1> vitoWiFi(&Optolink);

```
I´ve read all documentation, bug reports and your source code, but can´t find out how to use softwareserial in that case. It compiles fine, IR-LED is flashing, but all datapoints return timeouts. Maybe you can have a look please?

Thanks in advance!

bertmelis, 2024-09-22:
To be honest, I didn't test softwareserial with v3.
At first glance I don't see anything wrong with your code.

The softwareserial library might have changed their API. We might need to specify pins in the `begin` call instead of the constructor. I'll prepare a possible fix to test.

marcus2208, 2024-09-22:
Wow, that was a really fast response!

I don't think the pin assignment is a problem. The IR-LED is flashing, so the pin for TX at least should be right. But as I said, I'm a beginner in C and might be wrong.

bertmelis, 2024-09-23:
I actually don't have a clue at this point.

Would it be possible to enable debug logging? Are you using Arduino IDE or Platformio?

marcus2208, 2024-09-23:
I'm using Arduino IDE, latest 1.8.x
Debug logging... Either I have to take my notebook down to the basement, or I have to try it over telnet. First one seems easier, last one more comfortable. Right now I'm on the way to a customer, will try that at the weekend. Thank you for having a look into this.

bertmelis, 2024-09-23:
Could you test with this branch: https://github.com/bertmelis/VitoWiFi/tree/fix-softwareserial

marcus2208, 2024-09-23:
I'll try the fix-softwareserial as soon as I'm back home. Saturday or Sunday I think.

marcus2208, 2024-09-28:
I´ve just tried the fix-softwareserial branch. Sadly nothing changed, still all timeouts.
Now, for the debugging I need some instructions please. As far as I understand, I have to "#define DEBUG_VITOWIFI" and choose my debug port, e.g. "serial" in Arduino IDE? What about debug level in IDE? Leave it at "None"?

marcus2208, 2024-09-28:
Okay, I had #define DEBUG_VITOWIFI in my sketch. That was the fault I made. Debugging is working now.

```
16:49:03.029 -> reading packet OK
16:49:03.176 -> state 0 --> 2
16:49:03.176 -> state 2 --> 4
16:49:03.176 -> state 4 --> 5
16:49:07.042 -> state 5 --> 0
16:49:07.042 -> reading packet OK
16:49:07.390 -> state 0 --> 2
16:49:07.390 -> state 2 --> 4
16:49:07.390 -> state 4 --> 5
16:49:11.068 -> state 5 --> 0
16:49:11.068 -> reading packet OK
16:49:11.670 -> state 0 --> 2
16:49:11.670 -> state 2 --> 4
16:49:11.670 -> state 4 --> 5
16:49:15.028 -> state 5 --> 0
16:49:15.028 -> reading packet OK
16:49:15.885 -> state 0 --> 2
16:49:15.885 -> state 2 --> 4
16:49:15.885 -> state 4 --> 5
16:49:19.042 -> state 5 --> 0
16:49:19.042 -> reading packet OK
16:49:20.098 -> state 0 --> 2
16:49:20.098 -> state 2 --> 4
16:49:20.098 -> state 4 --> 5
16:49:23.055 -> state 5 --> 0
16:49:23.055 -> reading packet OK
16:49:24.359 -> state 0 --> 2
16:49:24.359 -> state 2 --> 4
16:49:24.359 -> state 4 --> 5
16:49:27.022 -> state 5 --> 0
16:49:27.022 -> reading packet OK
16:49:28.574 -> state 0 --> 2
16:49:28.574 -> state 2 --> 4
16:49:28.574 -> state 4 --> 5
16:49:31.029 -> state 5 --> 0
16:49:31.029 -> reading packet OK
```

bertmelis, 2024-09-28:
There might be a problem with the VS1 protocol implementation and not with your software serial.

Are you sure your heater doesn't work with VS2?

marcus2208, 2024-09-28:
I'm very sure it's using VS1 aka KW only. It's quite old. The boiler was installed in 2009, but the control unit was recycled from the old boiler, because it's compatible. I don't know exactly how old it is, because I bought the house in 2009. Vitosoft, the OEM software is speaking KW to this control unit.

bertmelis, 2024-09-28:
I fixed a mistake in the VS1 protocol. I think it should be better now. I pushed to the same branch.

marcus2208, 2024-09-28:
Sounds promising. Will try it later.

marcus2208, 2024-09-28:
Still no data
```
Datapoint "outsidetemp" error: timeout
reading packet OK
state 0 --> 1
state 1 --> 2
state 2 --> 4
state 4 --> 5
state 5 --> 0

```
Is state 3 missing also?

bertmelis, 2024-09-28:
Sorry, I'll analyze again tomorrow. I already turned off my computer.

marcus2208, 2024-09-28:
Of course. Have a good night!

bertmelis, 2024-09-28:
Did some reading. The timing is off with a factor 10 (500ms instead of 50). I will fix tomorrow.

marcus2208, 2024-09-28:
I tried to fix myself, but failed. 
In VS1.cpp, line 233:

`  } else if (_currentMillis - _lastMillis > 50) {
`

Sadly, still the same error. But probably thats not the only change is nessesary?
Will go to sleep now. Thanks for your support so far. Tomorrow we´re going to fix it, for sure. ;-)

bertmelis, 2024-09-30:
The weekend was busier than expected... I simplified the VS1 protocol implementation. You're the chosen one to test.

marcus2208, 2024-09-30:
Don't worry, weekend is weekend. Noone expected anything. I'm happy to test later, after daily business.

bertmelis, 2024-09-30:
Strange. You go through the whole loop so you must be reading the ACK from the connection.

I'm not at the computer right now. I'm sorry this takes so long.

bertmelis, 2024-10-01:
I reworked the protocol again.

Let's assume softwareserial isn't the problem.

marcus2208, 2024-10-01:
Nope, still the same. Here is the log, state 2 is missing now, I don´t know if it´s by intention:

```
15:48:47.559 -> state 0 --> 1
15:48:47.606 -> state 1 --> 0
15:48:49.768 -> state 0 --> 1
15:48:49.815 -> state 1 --> 0
15:48:52.015 -> state 0 --> 1
15:48:52.062 -> state 1 --> 0
15:48:54.269 -> state 0 --> 1
15:48:54.316 -> state 1 --> 0
15:48:56.521 -> state 0 --> 1
15:48:56.567 -> state 1 --> 0
15:48:57.272 -> reading packet OK
15:48:58.732 -> state 0 --> 1
15:48:58.732 -> state 1 --> 3
15:48:58.778 -> state 3 --> 4
15:49:01.270 -> state 4 --> 0
15:49:01.270 -> Datapoint "outsidetemp" error: timeout
15:49:01.270 -> reading packet OK
15:49:01.786 -> state 0 --> 1
15:49:01.786 -> state 1 --> 3
15:49:01.786 -> state 3 --> 4
15:49:05.267 -> state 4 --> 0
15:49:05.267 -> Datapoint "boilertemp" error: timeout
15:49:05.267 -> reading packet OK
15:49:06.254 -> state 0 --> 1
15:49:06.254 -> state 1 --> 3
15:49:06.254 -> state 3 --> 4
15:49:09.268 -> state 4 --> 0
15:49:09.268 -> Datapoint "watertemp" error: timeout
```

Do you have a Viessmann boiler? I think I´ve read it somewhere. Even if it´s VS2 protocol, it should talk VS1 also. This way we can confirm, the protocol is or isn´t the problem. And... Take your time. I´m not in hurry.

bertmelis, 2024-10-01:
Yes, state 2 is only when you read a second datapoint right after the first.

marcus2208, 2024-10-01:
I´m still not sure it´s the protocol. But let´s see. Please remember, it´s low priority. Again, thank you for your support!

bertmelis, 2024-10-01:
No worries. It sparked my interest. I'm flashing an SD for my raspberry pi to test.

bertmelis, 2024-10-02:
I think I got it working. Some issues with the serial interface on the raspberry pi though, but I don't think it propagates to the esp.

marcus2208, 2024-10-02:
Great job! 🥇 
It´s reading data now. No more timeouts! Thank you very much!

bertmelis, 2024-10-03:
another weight from my shoulders.

The branch is a bit messy now with all different changes. I'll clean up and merge soon.

bertmelis, 2024-10-03:
merged updates into `main`.

closed 2024-10-03 by bertmelis