bertmelis/VitoWiFi #119: Read and write schedule (enhancement)

Eugen4, 2024-12-24:
### Installation specifics
* **Heating type:** Vitotronic 200
* **Protocol:** P300
* **Board:** ESP8266 (Wemos D1 mini)

### Symptom
**What happens?** I wasn't able to read and write schedule for water heating
What would you expect? Add function to the library for simplify read and write schelude info to Viessmann

### Extra info
ESPHome provides such possibilities:
Schedule plans info: https://deploy-preview-2737--esphome.netlify.app/components/optolink.html#schedule-plans
Code reference: https://github.com/80morpheus/esphome_optolink/blob/optolink/esphome/components/optolink/helpers.cpp

bertmelis, 2024-12-24:
Can be implemented. Could you remind me in a week or so? I'm enjoying some time off 🎄.

bertmelis, 2025-01-07:
I don't use this feature myself so I'm not really familiar with the schedule.

A single schedule is encoded by a series of 8 bytes with each byte representing a time (rounded to 10 minutes). So for one day you can define 4 periods (time on + time off = 2 bytes). A whole week needs 7 * 8 bytes = 56 bytes, as referred to on the Home Assistant page.

I don't know if a single time entry can be read or written or the whole schedule (56 bytes) need to be requested.

So my main question: How would you use the schedule?

Eugen4, 2025-01-14:
Hello Bert. I would use schedule to control Water Heating. I can be away for days/weeks. In such cases I prefer to disable (before leaving house) water heating via schedule reconfiguration. I think it is a good posibility to remotelly turn on water heating just before comming back home.
Also I could define several presents in NodeMCU to upload them by single click.

bertmelis, 2025-01-14:
I understand that, but I don't know the datapoints myself. Are you writing 8 bytes in a single command or is it 56 bytes at once?

That said, I added already a preliminary version in the [`conv-schedule`](https://github.com/bertmelis/VitoWiFi/tree/conv-schedule)-branch.

No automatic conversion at the moment, only:
```
// "7:30 8:30 17:30 19:00" --> byte array
std::size_t VitoWiFi::encodeSchedule(const char* schedule, std::size_t len, uint8_t* output);
std::size_t VitoWiFi::encodeSchedule(const char* schedule, uint8_t* output);
// byte array --> "7:30 8:30 17:30 19:00"
std::size_t VitoWiFi::decodeSchedule(const uint8_t* data, std::size_t len, char* output, std::size_t maxLen);
```

Eugen4, 2025-01-14:
According to the HA I guess length should be 56 bytes.
I couldn't find exact initial address to write these 56 bytes. I think I could try to read data and compare with current configuration until correct address is found.

bertmelis, 2025-01-15:
I drafted some sample code (untested!).

```cpp
/*
a (weekly) schedule datapoint is 56 bytes:
7 days, 4 schedules per day with start and end time
--> 7 * 4 * 2 = 56 bytes

The conversion functions work with a daily schedule (8 bytes)
so we have to aggregate the daily schedules into a weekly and
vice versa.
*/

// define the datapoint
VitoWiFi::Datapoint() mySchedule("mySchedule", 0x1234, 56, VitoWiFi::noconv);

// print out the mySchedule, example for VS2
void onResponse(const VitoWiFi::PacketVS2& packet, const VitoWiFi::Datapoint& datapoint) {
  Serial.printf("%s:\n", datapoint.name());
  if (datapoint == mySchedule) {
    const std::size_t bufferLength = 48;
    char buffer[bufferLength];
    for (std::size_t i = 0; i < 56; i += 8) {
      if (decodeSchedule(&packet.data()[i], packet.dataLenght(), buffer, bufferLength) > 0) {
      Serial.printf("day %u: %s\n", i, buffer);
      } else {
        break;
      }
    }
  }
}

// schedule is always 7 days, 2 hardcoded schedules
const char* holidaySchedule[] = { "9:00 22:00",
                                  "9:00 22:00",
                                  "9:00 22:00",
                                  "9:00 22:00",
                                  "9:00 22:00",
                                  "9:00 22:00",
                                  "9:00 21:00" };
const char* workingSchedule[] = { "7:30 8:30 15:00 21:00",
                                  "7:30 8:30 15:00 21:00",
                                  "7:30 8:30 15:00 21:00",
                                  "7:30 8:30 15:00 21:00",
                                  "7:30 8:30 15:00 22:00",
                                  "9:00 22:00",
                                  "9:00 21:00" };

// encode the above schedule into a byte buffer
// and write the raw data to the datapoint
bool writeSchedule(const char** schedule) {
  std::size_t bufferLength = 56;
  uint8_t buffer[bufferLength];
  for (std::size_t i = 0; i < 8; ++i) {
     if (VitoWiFi::encodeSchedule(schedule[i], &buffer[i * 8]) == 0) {
       return false;
     }
  }
  return myVitoWiFi.write(mySchedule, buffer, bufferLength);
}

// usage:
writeSchedule(holidaySchedule);
*/
```

If you don't hardcode the schedule, a bit more formatting is needed. `encodeSchedule` expects a daily instead of weekly schedule. Depending on how you will send/receive the schedule string, more or less formatting is needed (eg split the strings based on line endings).

bertmelis, 2025-01-28:
Could you give some feedback on the functionality? Is it ok like it is? Would you like it more integrated?
Does it suit your need?

Eugen4, 2025-01-28:
Hello @bertmelis . Unfortunately, I had no chance to test it yet (I'm away right now). I will reply as soon as possible.

bertmelis, 2025-04-21:
Fixed with #126

closed 2025-04-21 by bertmelis