bertmelis/VitoWiFi #120: Build project problem

MomoFab1, 2025-01-03:
Thanks for sharing this work!

Install Problems:

- WIN10 VSCode Platformio Build not working for me:
	-pthread error (can be solved with -lpthread)
	g++ error: No such File or Dir... path problem,

I abondoned this way.

So I installed VSCode, Platformio on Linux Debian Bookworm,
not familiar with this OS I get error that I`m not able to solve:

 --------------------------------------------------------------------------------------------------------------------------------------------
 *  Executing task: platformio run --environment native 
Processing native (platform: native)
----------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- VitoWiFi @ 3.0.1
Building in debug mode
Compiling .pio/build/native/src/Datapoint/Converter.o
Compiling .pio/build/native/src/Datapoint/Datapoint.o
Compiling .pio/build/native/src/GWG/GWG.o
Compiling .pio/build/native/src/GWG/PacketGWG.o
Compiling .pio/build/native/src/Interface/HardwareSerialInterface.o
Compiling .pio/build/native/src/Interface/LinuxSerialInterface.o
Compiling .pio/build/native/src/Interface/SoftwareSerialInterface.o
Compiling .pio/build/native/src/VS1/PacketVS1.o
Compiling .pio/build/native/src/VS1/VS1.o
Compiling .pio/build/native/src/VS2/PacketVS2.o
Compiling .pio/build/native/src/VS2/ParserVS2.o
Compiling .pio/build/native/src/VS2/VS2.o
Compiling .pio/build/native/libce6/VitoWiFi/Datapoint/Converter.o
Compiling .pio/build/native/libce6/VitoWiFi/Datapoint/Datapoint.o
Compiling .pio/build/native/libce6/VitoWiFi/GWG/GWG.o
Compiling .pio/build/native/libce6/VitoWiFi/GWG/PacketGWG.o
Compiling .pio/build/native/libce6/VitoWiFi/Interface/HardwareSerialInterface.o
Compiling .pio/build/native/libce6/VitoWiFi/Interface/LinuxSerialInterface.o
Compiling .pio/build/native/libce6/VitoWiFi/Interface/SoftwareSerialInterface.o
Compiling .pio/build/native/libce6/VitoWiFi/VS1/PacketVS1.o
Compiling .pio/build/native/libce6/VitoWiFi/VS1/VS1.o
Compiling .pio/build/native/libce6/VitoWiFi/VS2/PacketVS2.o
Compiling .pio/build/native/libce6/VitoWiFi/VS2/ParserVS2.o
Compiling .pio/build/native/libce6/VitoWiFi/VS2/VS2.o
Archiving .pio/build/native/libce6/libVitoWiFi.a
Indexing .pio/build/native/libce6/libVitoWiFi.a
Linking .pio/build/native/program

/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/12/../../../x86_64-linux-gnu/Scrt1.o: in function `_start':
(.text+0x17): undefined reference to `main'

collect2: error: ld returned 1 exit status
*** [.pio/build/native/program] Error 1

Please, how to solve it.

Happy New Year!!!

bertmelis, 2025-01-03:
I think you are trying to compile the library as a program.

You should compile one of the examples (create your own based on the examples) and use the library.

MomoFab1, 2025-01-03:
yes, You are right, because as a 60yo noob I followed and execute your install instractions [here ](https://registry.platformio.org/libraries/bertmelis/VitoWiFi/installation)step by step:

  Depend on it
1. Open [platformio.ini](https://docs.platformio.org/en/latest/projectconf/index.html), a project configuration file located in the root of PlatformIO project.
2. Add the following line to the [lib_deps](https://docs.platformio.org/en/latest/projectconf/sections/env/options/library/lib_deps.html) option of [env:] section:
bertmelis/VitoWiFi@^3.0.1
3. **### Build a project, PlatformIO will automatically install dependencies.**

However,
compile a program... how?

Okay, I understand, there are basics on how to use platformio.
I can learn it the hard way or you are so kind to help me out.

Would you?

closed 2025-01-03 by MomoFab1

bertmelis, 2025-01-03:
Are you using linux natively or via WSL? I'm asking because this library uses serial communication. This is (imho) rather complicated in WSL and requires extra steps.

Do these steps for native development.

1. (`sudo apt-get update`)
2. install build tools, which platformio doesn't do for you for native development: `sodu apt-get install build-essential`
3. create a new platformio project, use native
4. add the library as a dependency like you did
5. copy the sample code of the "linux" example to main.cpp in the "src" directory. you might also want to copy the contents of the included platformio.ini into your own ini-file.

Now you should be able to compile the program.