bertmelis/VitoWiFi #123: Setting port-pins

KLelong, 2025-02-15:
Hi,
A while ago I was experimenting with the V1-software on an ESP32-wroom-board. Worked fine.
Now I flashed an example of the V3-software, but it seemed there is no communication on the serial TX-pin.
So I wondered what pins are used. In V1, you had to set the pins with e.g.   `VitoWiFi.setup(&Serial2, 16, 17); ` 
I then modified the example : `#define SERIAL1 Serial2` so it uses Serial2. Now the is some communication on the Tx2-pin  of the board.
Since those serial pins can be located (almost) anywhere, it would be nice to be able to define which pins to use.

I have to mount my hardware on my heater to see if it really works.

bertmelis, 2025-02-16:
I didn't implement this feature in VitoWiFi itself.

You can however do this in your code because you have access to all the Serial objects. Just call Serial.setPins(...) and it should work.

bertmelis, 2025-04-21:
Try something like this:

```cpp
VitoWiFi::VitoWiFi<VitoWiFi::VS2> vitoWiFi(&Serial1);

void setup() {
  Serial1.setPins(RXPIN, TXPIN);
  vitoWiFi.begin();
}
```

closed 2025-04-21 by bertmelis