bertmelis/VitoWiFi #130: Missing methods

s0170071, 2025-04-21:
In my sketch I do read DPs all the time and publisch them to mqtt. 
I also have a web interface that allows me to quick and dirty access DPs: 

[image]

However, this requires a method of determining if the interface is available.  I would love these methods: 

myHeater->isBusy()
myHeater->readRaw (word address, byte Len, uint8_t* rawdata )
myHeater->writeRaw ((word address, byte Len, uint8_t* rawdata )

bertmelis, 2025-04-21:
Are you using VS2?

I can implement in VS2 and if OK, expand the functionality to the other versions.

bertmelis, 2025-04-21:
The definition of *busy* is a bit grey though. What would you describe as being busy?
Does it mean "not ready to accept a new read or write"? But that would also include setting up the connecting.

Follow implementation in https://github.com/bertmelis/VitoWiFi/tree/raw_comms

bertmelis, 2025-04-21:
Mind that the `read` and `write` methods also return `bool`, indicating the same thing. The `isBusy` method is therefore a bit redundant.

bertmelis, 2025-04-21:
- `readRaw (word address, byte Len, uint8_t* rawdata )`
what is the purpose of `uint8_t* rawdata`? when reading, there is no data to be added.

- `writeRaw ((word address, byte Len, uint8_t* rawdata )` is already somewhat implemented: 
`write(Datapoint datapoint, const uint8_t* data, uint8_t length)`

You can construct a datapoint:
```cpp
VitoWiFi::Datapoint temporaryDP("noname", address, length, VitoWiFi::noconv);
```

You can use that to write using the method above.
If I'd to implement it in the library, I would do the very same thing. Or would you like to have an API to write arbitrary data to optolink? That is: regardless of the state, write any data and not have the lib calculate the checksum etcetera.

closed 2025-04-21 by bertmelis