bertmelis/VitoWiFi #131: The write method seems not working

lolo233, 2025-04-21:
Hi,

I have upgrade from VitoWifi V1 (was using it since a couple of years, thx !) to the V3.

The write method seems not OK.
I receive a read message with a different value, and the value of the data point is not set.

Any idea ?

bertmelis, 2025-04-21:
Looking back at the code, writing is totally flawed I think. I will come up with a solution!

bertmelis, 2025-04-21:
I closed the other issue as I was investigating this and spotted the very same things you did. Somethings seriously went wrong with that part of the code...

It compiles but I currently can't "live" test this myself. Could you give me feedback?

lolo233, 2025-04-21:
Hey, thank you so much for the fast fix !

It works, values are successfully written.
However the call to the write method triggers also the onError callback, but the rest looks OK.

bertmelis, 2025-04-21:
What is the error code?

closed 2025-04-21 by bertmelis

lolo233, 2025-04-21:
VitoWiFi::OptolinkResult::ERROR