bertmelis/VitoWiFi #134: malloc() / free()

s0170071, 2025-04-21:
Bert, I may be wrong, but as I was trying to figure out whats going on in 

```
bool VS2::write(const Datapoint& datapoint, const VariantValue& value) {
  if (_currentDatapoint) {
    return false;
  }
  uint8_t* payload = reinterpret_cast<uint8_t*>(malloc(datapoint.length()));
  if (!payload) return false;
  _currentDatapoint.encode(payload, _currentDatapoint.length(), value);
  return write(datapoint, payload, _currentDatapoint.length());
}
```

I got some questions that I am not so sure about. 

1. the function uses malloc() but I never see free(). Since malloc puts stuff on the heap, it should remain allocated, even when the function exits.
2. `_currentDatapoint.encode` is called and yet `datapoint `is written in the following line. 
3.   'const VariantValue& value' is dark arts, indeed. But I like it. At least in the Arduino IDE its giving me a hard time to use in my own code: 

I can do
void writeDP (const VitoWiFi::Datapoint VitoWifiDatapoint,  unsigned char value){
           while (! myHeater.write (VitoWifiDatapoint,     value )){  
             delay(10);
             myHeater.loop();
          }
}

Can't do ('VariantValue' does not name a type) : 
```
void writeDP (const VitoWiFi::Datapoint VitoWifiDatapoint,  const VariantValue& value){
           while (! myHeater.write (VitoWifiDatapoint,     value )){  
             delay(10);
             myHeater.loop();
          }
}
```
I did include <VitoWiFi.h> so maybe that ArduinoIDE specific behaviour.

Btw, Thanks for your swift replies, I really appreciate it.

bertmelis, 2025-04-21:
I was a bit eager to close this issue since it isn't really a duplicate nor is it fixed already.

anyway, the fix is included in #133 .

closed 2025-04-21 by bertmelis

bertmelis, 2025-04-21:
Could you point me to potential memory leaks?
Apart from the ones just fixed, I believe the buffers are freed on destruction.