bertmelis/VitoWiFi #135: onError called on successful write

bertmelis, 2025-04-21:
> Hey, thank you so much for the fast fix !
> 
> It works, values are successfully written.
> However the call to the write method triggers also the onError callback, but the rest looks OK. 

 _Originally posted by @lolo233 in [#131](https://github.com/bertmelis/VitoWiFi/issues/131#issuecomment-2819001449)_

bertmelis, 2025-04-22:
The parser doesn't parse correctly. It assumes the returned message after a write is the same message, including the data. But the data isn't returned.

The mistake is made because the openv wiki is wrong: https://github.com/openv/openv/wiki/Protokoll-300
The example is different from the explanation.

lolo233, 2025-04-22:
I have read something similar, that confirms there is no data in the response and then error during parsing. It means the interface needs to maintain the last message sent, or the response allows to identify there is no data (simpler to fix in this case, one condition to add). I can easily run a test on a beta version if you need.

bertmelis, 2025-04-23:
We can have either way. Or the library remembers the data and on success it is returned with the response. Or the response is just the "empty" response.

In case of the latter, I won't just return because you will get a packet with a certain data length but without data. This can be confusing. I'd just set the data length to zero.

Your call but I would remember the data. Easier for integrating into other systems I think.

closed 2025-04-24 by bertmelis

lolo233, 2025-04-25:
I installed the main branch today, this version triggers a crash after the write call (data are received by the heater). I guess this a memory exception when using a null pointer. I have commented the following instructions and it seems OK, but cannot say if it is the right solution.

`const uint8_t* PacketVS2::data() const {
  /* if (functionCode() == FunctionCode::WRITE) return nullptr; */
  return &_buffer[6];
}`

bertmelis, 2025-04-25:
Yeah, the library doesn't remember the written data (yet). The data method returns nullptr although length is not zero.

Returning something with the code as is is obviously not good. Nobody knows what's in that memory.

lolo233, 2025-04-25:
I confirm, with the commented instruction, the read callback is trigger and gives a wrong value of the datapoint, but then no crash of the ESP. I keep this "patched version" that allow to change the value and do not crash. I do a read read after the write, then the wrong dummy value I got from the write callback is overiden with this read. That works :)

PR #141 (2025-04-24): Fix onError on write response
Closes #135