bertmelis/VitoWiFi #140: str_to_raw function

s0170071, 2025-04-22:
When I receive values over MQTT, they come as strings.  What is does is accepting a string, e.g.  "23.4" and convert it into raw to send over the interface. What is the modern equivalent of that ?

```
void StrToRaw (String value, String type, uint8* raw ){
  
      for (int i = 0; i<10; i++) raw[i]=0;
      if (type == "TempL")      {   DPTemp temp_dp("TempL", "DPs", 0x00);       DPValue tempDP((float)atof(value.c_str())); temp_dp.encode(raw,tempDP);    } 
      else if (type == "TempS") {   DPTempS temps_dp("TempS", "DPs", 0x00);     DPValue tempDP((uint8_t)atoi(value.c_str())); temps_dp.encode(raw,tempDP);   }
      else if (type == "STAT")  {   DPStat stat_dp("Stat", "DPs", 0x00);        DPValue tempDP((bool )atoi(value.c_str())); stat_dp.encode(raw,tempDP);    }
      else if (type == "CountL"){   DPCount count_dp("CountL", "DPs", 0x00);    DPValue tempDP((uint32_t)atoi(value.c_str())); count_dp.encode(raw,tempDP);   }
      else if (type == "CountS"){   DPCountS counts_dp("CountS", "DPs", 0x00);  DPValue tempDP((uint16_t)atoi(value.c_str())); counts_dp.encode(raw,tempDP);  }
      else if (type == "Mode")  {   DPMode mode_dp("Mode", "DPs", 0x00);        DPValue tempDP((uint8_t)atoi(value.c_str())); mode_dp.encode(raw,tempDP);    }
      else if (type == "Hours") {   DPHours hours_dp("Hours", "DPs", 0x00);     DPValue tempDP((uint16_t)atoi(value.c_str())); hours_dp.encode(raw,tempDP);   }
      else if (type == "CoP")   {   DPCoP cop_dp("CoP", "DPs", 0x00);           DPValue tempDP((uint8_t)atoi(value.c_str())); cop_dp.encode(raw,tempDP);     }
}
```

bertmelis, 2025-04-22:
Use any of the numeric conversion functions from <string>, see: https://en.cppreference.com/w/cpp/string/basic_string

In V3 of VitoWiFi, you can then convert it automatically using VariantValue. This is a class that has constructors for the common types.

s0170071, 2025-04-22:
Thanks. I figured I can use variant to encode any type to a buffer:

```
uint8_t TypeStringToLen (String type){
  byte readLen=4;   // for  CountL, Hours and Raw
      if      ( (type == "TempL") || (type == "CountS"))       readLen = 2;
      else if ( (type == "STAT" ) || (type == "TempS")  || (type == "Mode") || (type == "CoP")|| (type == "Power") ) readLen = 1;
  return readLen;
}

void StrToRaw (String value, String type, uint8* raw ){
      for (int i = 0; i<10; i++) raw[i]=0;  
      const VitoWiFi::Datapoint dp("dummy",0x0000, TypeStringToLen(type), VitoWiFi::noconv );
      if (type == "TempL")      { dp.encode(raw, 10, VitoWiFi::VariantValue((float)   atof(value.c_str()))); } 
      else if (type == "TempS") { dp.encode(raw, 10, VitoWiFi::VariantValue((uint8_t) atoi(value.c_str()))); }
      else if (type == "STAT")  { dp.encode(raw, 10, VitoWiFi::VariantValue((bool)    atoi(value.c_str()))); }
      else if (type == "CountL"){ dp.encode(raw, 10, VitoWiFi::VariantValue((uint32_t)atoi(value.c_str()))); }
      else if (type == "CountS"){ dp.encode(raw, 10, VitoWiFi::VariantValue((uint16_t)atoi(value.c_str()))); }
      else if (type == "Mode")  { dp.encode(raw, 10, VitoWiFi::VariantValue((uint8_t) atoi(value.c_str()))); }
      else if (type == "Hours") { dp.encode(raw, 10, VitoWiFi::VariantValue((uint16_t)atoi(value.c_str()))); }
      else if (type == "CoP")   { dp.encode(raw, 10, VitoWiFi::VariantValue((uint8_t) atoi(value.c_str()))); }
}

```

But how does the conversion know how many bytes the raw data should use ? Does it use the len parameter of the Datapoint ? 
The address is not used here, right ?

s0170071, 2025-04-23:
And the RawToStr counterpart:

```
enum EnumDataType { na, CountS, CountL, Mode, TempS,TempL,Hours,CoP,STAT,Power,Raw, all};

String RawToStr (EnumDataType type, uint8* raw, uint8 readLen=0,  double factor=0){
        
      String retval; 
      String retval_all;
      readLen = TypeStringToLen(type);
      if (readLen>10) {//addLog(LOG_LEVEL_ERROR,"Readlen>10");  
          readLen=10;}

      if (type == TempL){
          const VitoWiFi::Datapoint dp("dummy",0x0000, TypeStringToLen(type), VitoWiFi::div10 );
          VitoWiFi::VariantValue result((float) 0.0); 
          result = dp.decode (raw, readLen);
          return (String ((float) result));
           
      } 
      else if ((type == STAT) || (type == TempS)||(type == Mode)){
          const VitoWiFi::Datapoint dp("dummy",0x0000, TypeStringToLen(type), VitoWiFi::noconv );
          VitoWiFi::VariantValue result((uint8_t) 0); 
          result = dp.decode (raw, readLen);
          return (String ((uint8_t) result));
      }
      else if (type == CountL){
          const VitoWiFi::Datapoint dp("dummy",0x0000, TypeStringToLen(type), VitoWiFi::noconv );
          VitoWiFi::VariantValue result((uint32_t) 0); 
          result = dp.decode (raw, readLen);
          return (String ((uint32_t) result));
      }
      else if (type == CountS){
          const VitoWiFi::Datapoint dp("dummy",0x0000, TypeStringToLen(type), VitoWiFi::noconv );
          VitoWiFi::VariantValue result((uint16_t) 0); 
          result = dp.decode (raw, readLen);
          return (String ((uint16_t) result));
      }
      else if (type == Hours){
          const VitoWiFi::Datapoint dp("dummy",0x0000, TypeStringToLen(type), VitoWiFi::div3600 );
          VitoWiFi::VariantValue result((float) 0.0); 
          result = dp.decode (raw, readLen);
          return (String ((float) result));
      }
      else if (type == CoP){
          const VitoWiFi::Datapoint dp("dummy",0x0000, TypeStringToLen(type), VitoWiFi::div10 );
          VitoWiFi::VariantValue result((float) 0.0); 
          result = dp.decode (raw, readLen);
          return (String ((float) result));
      }
      
      else if (type == Raw)   {
        retval="";
        for (int i = 0; i<readLen; i++) {
          if (raw[i] < 0x10) retval +="0"; 
          retval += String(raw[i],HEX);
        }
      }
      else if (type == all) {
        retval_all = "TempL:  " + RawToStr(TempL,raw,0,factor) +"\n";
        retval_all += "TempS:  " + RawToStr(TempS,raw,0,factor) +"\n";
        retval_all += "STAT:   " + RawToStr(STAT,raw,0,factor) +"\n";
        retval_all += "CountL: " + RawToStr(CountL,raw,0,factor) +"\n";
        retval_all += "CountS: " + RawToStr(CountS,raw,0,factor) +"\n";
        retval_all += "Mode:   " + RawToStr(Mode,raw,0,factor) +"\n";
        retval_all += "Hours:  " + RawToStr(Hours,raw,0,factor) +"\n";
        retval_all += "CoP:    " + RawToStr(CoP,raw,0,factor) +"\n";
        retval_all += "Raw:    " + RawToStr(Raw,raw, readLen ) +"\n";
        return retval_all;
      }
      
      else  retval = "unknown encoding requested";
      return retval;
 
}
```