bertmelis/VitoWiFi #146: Questions about CRC-errors

KLelong, 2025-05-29:
Hi,
I'm testing vitowifi with my heater. After establishing the type, I'm now testing what datapoints I can use, and how to do it.
I can get a number of datapoints, but one generates a CRC error. Is that error really a checksum-error or can it be triggered by another error in the communication ?
Can the software recover from a crc-error ? A first loop through my datapoints gets good values, but the last datapoint is the one that generates the crc-error. Unfortunately, all datapoints after that result in crc-errors. See the screenshot of the serial-output.
FWIW this is the definition of my datapoints : 
```
VitoWiFi::Datapoint datapoints[] = {
  VitoWiFi::Datapoint("deviceID1", 0x00F8, 4, VitoWiFi::noconv),
  VitoWiFi::Datapoint("outsidetemp", 0x5525, 2, VitoWiFi::div10),
  VitoWiFi::Datapoint("boilertemp", 0x0810, 2, VitoWiFi::div10),
  VitoWiFi::Datapoint("boilertemp set", 0x555A, 2, VitoWiFi::div10),
  VitoWiFi::Datapoint("burnerstarts", 0x088A, 4, VitoWiFi::noconv),
  VitoWiFi::Datapoint("boilerhours", 0x08A7, 4, VitoWiFi::div3600),
  VitoWiFi::Datapoint("A1M1 roomtemp", 0x0896, 2, VitoWiFi::div10),
  VitoWiFi::Datapoint("A1M1 supply temp", 0x0810, 2, VitoWiFi::div10),
  VitoWiFi::Datapoint("burner", 0x55d3, 1, VitoWiFi::noconv),   
  VitoWiFi::Datapoint("A1M1 actual mode", 0x55d3, 1, VitoWiFi::noconv),
  VitoWiFi::Datapoint("A1M1 party", 0x2303, 1, VitoWiFi::noconv),
  VitoWiFi::Datapoint("A1M1 save", 0x2302, 1, VitoWiFi::noconv),
  VitoWiFi::Datapoint("pump", 0x7760, 1, VitoWiFi::noconv)
  
//  VitoWiFi::Datapoint("deviceID2", 0x00FC, 4, VitoWiFi::noconv),
};
```
This is an extract of the datapoint-definition : 
```
VScot HO1
=========

# Überblick (8854)
- Kessel (6840)
    - Aussentemperatur (5373) [TiefpassTemperaturwert_ATS~0x5525 (SInt)]
    - Kesseltemperatur (5374) [TiefpassTemperaturwert_KTS~0x0810 (Int)]
    - Brenner (600) [GWG_Flamme~0x55D3 (Byte)]
    - Kesselsolltemperatur (2954) [Kesselsoll_eff~0x555A (Byte)]
    - Interne Pumpe Drehzahl (787) [InternePumpeDrehzahl~0x7660 (Int)] HIDDEN:(30_KennIntUmwPumpe="0 stufig")
    - Interne Pumpe (245) [DigitalAusgang_InternePumpe~0x7660 (Byte)]
    - Brennerstarts (111) [BrennerstartsGWG~0x088A (Int4)]
    - Brenner-Betriebsstunden (104) [BetriebsstundenBrennerBedienungGWG~0x08A7 (Int)]
    - Hydraulische Weiche (8161) [Temperatur_Hydraulische_Weiche~0x080C (Int)] HIDDEN:("(52) Sensor Hydraulische Weiche"="0 nicht vorhanden")
- Heizkreis A1 (6841) HIDDEN:(00_WWS="3 M2" OR 00_WWS="4 M2 +  WW")
    - Aktuelle Betriebsart A1M1 (708) [HK_AktuelleBetriebsartA1M1~0x2500 (Byte)]
    - Externe Betriebsarten-umschaltung A1M1 (270) [ExtBetriebsartenumschaltung_A1M1~0x2549 (Byte)] HIDDEN:(ZuordExtBetrUmsch="0 inaktiv" OR ZuordExtBetrUmsch="2 M2" OR ZuordExtBetrUmsch="4 M3" OR ZuordExtBetrUmsch="6 M2 + M3")
    - Vorlauftemperatur A1M1 (5375) [TiefpassTemperaturwert_KTS_A1~0x0810 (Int)]
    - Heizkreispumpe A1 (39) [Ausgang_HKP_A1~0x7663 (Byte)]
    - Heizkreispumpe A1M1 Drehzahl (778) [HKP_A1Drehzahl~0x7663 (Int)] HIDDEN:(E5_KennPumpHzkA1="0 stufig")
    - Partybetrieb A1M1 (76) [BedienPartybetriebA1M1~0x2303 (Byte)]
    - Sparbetrieb A1M1 (88) [BedienSparbetriebA1M1~0x2302 (Byte)]
    - Raumtemperatur A1M1 (5367) [TiefpassTemperatur_RTS_A1M1~0x0896 (Int)] HIDDEN:(FBA1M1="nicht vorhanden" OR A0_KennFBA1M1="0 ohne" OR A0_KennFBA1M1="nicht vorhanden" OR RaumTempSensHK1="Nicht vorhanden")
```
[image]

bertmelis, 2025-05-30:
I think I found the culprit:

Add "_step = ParserStep::STARTBYTE" before line 107 in ParserVS2.cpp

The parser doesn't reset if a CRC error is detected.

bertmelis, 2025-06-02:
The parser should now recover from CRC errors.

Please reopen if the issue persists.

closed 2025-06-02 by bertmelis

KLelong, 2025-06-02:
By making a mistake I discovered some bugs ;-) . First, that CRC-error, now another one I think.
I asked for a non-existing datapoint (0x7760, instead of 0x7660). So the heater responded with an error (I think). It seems that that error is not processed (well).
I "sniffed" the serial line (all hex values, TX is to heater) : 
```
TX : 06 41 05 00 01 77 60 01 de
RX : 06 41 06 03 01 77 60 01 01 e3 06
```
Simultanious with that e3, there is an 04 sent to the heater, if that matters.
I tried using -DDEBUG_VITOWIFI=1 in my platformio-file to have some debugging-info, but on the serial port where the responses are show, nothing gets written. Where does that `vw_log_w` go ?